mikeakohn/naken_asm #3: Incorrect label address with mov on STM8

abe-k, 2015-03-25:
If a `mov` instruction needs to use the long addressing mode, its length will change, but this mistakenly does not update label addresses. For example,
```
.stm8
.org 0x100
mov test, test
test: .ds8 1
```
produces the listing
```
0x0100: 55 01 03 01 03 mov $103, $103
```
when it should produce
```
0x0100: 55 01 05 01 05 mov $105, $105
```
since the `mov` instruction is now 5 bytes long.

mikeakohn, 2015-04-14:
Sorry for not replying earlier.. I didn't see this message until today.  I'm looking into it and hope to have it fixed by tomorrow night. Thanks for letting me know.

abe-k, 2015-04-14:
An interesting corner case to keep in mind, that some of the obvious approaches will fail on, is
```
.stm8
.org 0x00
mov 0x103 - test, 0x103 - test
test: .ds8 1
```
If `mov` is three bytes long, then `test = 0x03`, so `0x103 - test = 0x100`, so `mov` needs to be five bytes long, which makes `test = 0x05` and so `0x103 - test = 0xFE`, but this doesn't allow reducing `mov` back to three bytes or you get an infinite loop.

mikeakohn, 2015-04-16:
Should be fixed.  I tested your second case too and it seems to work.  All the old automated tests pass.

I was adding automated tests that catch this issue on 68000... I need to add the same automated tests for STM8 so all forms of the instructions are tested and this doesn't happen again.

Thanks for being patient.  Let me know if you find anything else.

closed 2015-04-16 by mikeakohn