mikeakohn/naken_asm #4: Short addressing forms not used with labels on STM8

abe-k, 2015-04-22:
Many of the STM8 instructions have different forms for one-byte and two-byte addresses, and the one-byte form is preferable for space efficiency whenever possible. However, when an address argument depends on a label that has not yet been defined, the assembler currently always chooses the long form, even when the argument could fit in a byte. I have checked that this occurs with `mov`, `ld`, `ldw`, and `neg`, so it probably applies to most instructions. Here is an example:
````
.stm8
.org 0x00
ld a, test
test: .ds8 1
````
produces
````
0x0000: c6 00 03  ld A, $3
0003: 00
````
when it would be better to produce
````
0x0000: b6 02  ld A, $02
0002: 00
````
It seems that this issue is partly caused by the fix to issue #3, as prior to that `mov` would attempt to use the short form.

mikeakohn, 2015-04-27:
naken_asm always picks the long form of an instruction if it can't resolve a symbol on pass 1.  Try defining your data at the top of the file instead of the bottom and see if it still picks the long form.

abe-k, 2015-04-27:
This isn't a big issue, true; assuming no backwards movement with `.org`, this will only come up when putting code in the the first page of RAM, which is unusual. I don't think this is at all priority, but it would be nice for completeness.

mikeakohn, 2015-04-27:
I looked into this with other assemblers a while ago... other assemblers were doing the same as naken_asm with this issue.  I'm not sure what ST's assembler does, but I can try it.

Btw, github's notifications to my email seem to be getting filtered it.  Feel free to ping me by email if I don't respond your issue requests.

abe-k, 2015-04-27:
It's perfectly reasonable not to shorten forward references, since it seems it would require at least an extra pass.

Of the current issues, only #6 is crucial for my current project. I'll email you if I post any more.

Thanks, by the way, for creating this—the only other open assembler I've found for STM8 is the ASxxxx assembler in SDCC, which I've had some problems with.

mikeakohn, 2015-04-27:
I fixed the problem with github emails so it should be good now.

I'm working on your other two issues now and will go ahead and close this one for now.  And I need to send thanks back for letting me know about these issues.  I wrote the STM8 module for the assembler and never really had time to use it myself (was having trouble trying to flash a chip using Linux)... so by using it and reporting issues to me you've been a big help.

closed 2015-04-27 by mikeakohn