mikeakohn/naken_asm #5: Forward label references not supported in indexed or indirect addressing modes on STM8

abe-k, 2015-04-26:
The assembler currently fails to understand instructions that use indexed or indirect addressing modes with a label that is yet to be defined. For example,
````
.stm8
.org 0
ld a, (test, x)
test: .ds8 1
````
fails with

````
Error: Unknown operands combo for 'ld' at filename:3.
````
while moving the label definition before the instruction avoids this:
````
.stm8
.org 0
test: .ds8 1
ld a, (test, x)
````
succeeds. In indirect addressing modes, it fails at the parsing stage:
````
.stm8
.org 0
ld a, ([test], x)
test: .ds 1
````
fails with
````
Error: Unexpected token ',' at filename:3
````
This is specific to a subset of addressing modes (I have not tested all of them), as
````
.stm8
.org 0
ld a, test
test: .ds8 1
````
assembles (though see issue #4).

mikeakohn, 2015-04-27:
I'll look into this one.

mikeakohn, 2015-04-28:
Should be fixed now.

closed 2015-04-28 by mikeakohn