mikeakohn/naken_asm #7: Feature request: Local labels

abe-k, 2015-04-28:
Within an assembly function, there are often many internal jumps for conditionals and loops. The labels for these jumps do not need to be visible outside the scope of the function. It would be good to have some form of local label, which is only visible within some context, and which does not need a globally unique name.

Usually, some feature of the symbol name distinguishes local from global labels: some assemblers have local labels as those that begin with a `.`, some those that begin with a digit, and some those that end with a `$`. My personal preference is for labels beginning with `_` to be considered local.

There are also a few ways to define the context for a local label: some assemblers consider the context to be the area between two non-local labels, some require explicit directives to specify the area, and some require specifying when using the label whether to search forward or backward.

While of course it is easy to fake having local labels—say, by prepending an underscore and the function name—it makes for slower-to-type and harder-to-maintain code than does proper support.

Thoughts on usefulness or on specifics of syntax and semantics?

mikeakohn, 2015-04-29:
I've had similar requests in the past for things like this.  I think the _ to start a local label would be out for me since I use _label in programs like Java Grinder.  The .label wouldn't be so bad I think.  I'm not sure how easy this would be to implement though.  I have to think about it.

So if there was:

label1:
  mov.w r4, r5
.loop:
  add.w r5, r6
  jnz .loop

label2:

Then, when the assembler sees label1:, it becomes the current label.  When it sees .loop it will record it as label1.loop and when it's used it will use it as label1.loop.  If .loop is referred to under label2:, the assembler would assume it's label2.loop, not find it, and throw an error.

Actually, I'm not sure . could easily be used because of the way the tokenizer works.

Have to think about it more.

abe-k, 2015-04-29:
The disadvantage of using `.`, even if it's unambiguously tokenizable, is that it would be easy to confuse with a directive—imagine having a local label `.endif`. The assembler so far only seems to use `$` as an alternate notation for hexadecimals, so perhaps it would be a clearer choice, if `_` is out for compatibility reasons.

Using nonlocal labels as context for local ones does seem the most straightforward. Of course, you'll want to make sure the internal combined representation can't conflict with a real nonlocal label.

mikeakohn, 2015-09-01:
Just thought of something.  What if there was a directive such as:

.function blah

loop:
  jnz loop
.endf

any labels created between .function blah and .endf would only be valid in the function.. in this case loop couldn't be used outside of that block of code.

abe-k, 2015-09-01:
That seems like a good approach. I'd additionally suggest that `.function blah` have the effect of `blah:`, that is, as well as defining a scope it creates the label for that function.

ofosos, 2015-09-13:
I would like to use local labels in a macro. .function seems inappropriate for this. Maybe just implement something like .scope?

mikeakohn, 2015-10-31:
Sorry I haven't implemented this feature yet.  Been really busy.  Maybe the clean way would be:
.local name:

instead of just:

name:

If it's inside a macro or inside a function, the label will disappear at the end of the macro or function.

abe-k, 2015-11-01:
I don't like the idea of having to type `.local name:` for every local label; that's as much typing as prefixing each local label with the name of the function it's in, as I've been doing. I'd much prefer `.function` (or `.scope`) to cause all labels declared inside it to automatically be local.

mikeakohn, 2015-11-01:
Yeah.. the local thing is ugly.  I have another idea.  How about a .delete or .delete_label thing (not sure what keyword is more obvious:

blah:
  ...
  jnz blah
.delete blah

I could do it all three ways.  .function always scopes (or if it makes more sense .funct .proc etc.. not sure which keyword is more common), .macro always scopes, and .delete always removes the label.

abe-k, 2015-11-01:
If `.function` and `.macro` both scope, I don't see the need for `.delete`. You're unlikely to have either code outside a function/macro or a function/macro so large you want to reuse labels within it.

It would probably be good to abbreviate to `.func`, so that `.endfunc` can match without being the unwieldy `.endfunction` (though you do have `.endm` instead of `.endmacro`, so `.endf` would match either).

I guess the remaining question is, is there a need for `.scope` in addition, that only creates a scope without defining a macro or declaring a label? I don't see one offhand, but I don't see it hurting either.

One thing to watch out for in implementation is that labels declared in a macro need to be local to each instantiation of the macro, rather than merely local to the macro definition.

hosewiejacke, 2016-04-30:
In golang the visibility depends on the case of the 1st letter:
* `MyFunc` is visible to all,
* `myFunc` is visible only in the current package.

I know this seems really strange at first...

mikeakohn, 2016-05-03:
Okay.  I added .func / .endf and .scope / .ends.  The only negative is it doesn't support nested scope.  I can add it in later if it's really needed, but for now this works.  .scope can be used in macros to paste them multiple times.  Actually, it's limited to 65535 scopes.. maybe that needs to be raised to 4 billion.  Anyway, sample code:

`main:
  mov.w #400, r4
  call #junk
.scope
repeat:
  dec.w r4
  jnz repeat
.ends
  mov.w #200, r4
  call #junk
repeat:
  jmp repeat

.func junk
repeat:
  dec.w r4
  jnz repeat
  ret
.endf
`

hosewiejacke, 2016-05-04:
Great, thanks!

hosewiejacke, 2016-06-02:
I think this is resolved, isn't it?

closed 2016-06-02 by mikeakohn