mikeakohn/naken_asm #10: STM8 disassembler skips most bytes

abe-k, 2015-06-25:
The disassembler for the STM8 architecture does not take into account the variable instruction length. It tries to decode one instruction at every four-byte boundary, even though instructions can be anywhere from one to five bytes long. Rather than always advancing `start` by 4, it should advance it by however long the previously decoded instruction was.

Additionally, the display is not accurate: rather than showing the bytes that make up the instruction, it shows the two bytes at the beginning of the four-byte word, padded with two bytes worth of zeros.

mikeakohn, 2015-06-25:
Don't have time to look at this and the other two issues now.  I'll get back to this in about 12 hours.

mikeakohn, 2015-06-26:
Should be fixed now.

closed 2015-06-26 by mikeakohn

abe-k, 2015-06-26:
The disassembly works great now, but the output is still showing two bytes for the opcode (`num`) regardless of instruction length, and still always padded to four bytes.