mikeakohn/naken_asm #12: Request: separate error output

abe-k, 2015-06-25:
Currently, both standard messages (such as version info, include paths, and code size) and error and warning messages are printed to `stdout`. It would be convenient if the errors and warnings were printed to `stderr` instead, so that one could suppress `stdout` (i.e. `> /dev/null`) to make it easier to see whether assembly had succeeded and what had gone wrong. Alternately, a command line "quiet" option could be added, that would suppress all messages except errors and warnings internally.

mikeakohn, 2015-07-26:
I added a -q option to limit output.  It still prints the credits.. I was going to quiet that also, but that might take a little work since.. well, I was going to move the credits under the command line argument part, but the output could get awkward that way.  Let me know if this is good enough for you and I'll close it, otherwise I'll add the extra code to hide the credits too.

abe-k, 2015-07-26:
It's easy enough from a Makefile to hide the credits by piping to `tail`, since they're a fixed number of lines at the beginning. Ideally, of course, `-q` would hide everything except error messages, but I get how that's a bit of a pain when printing notifications before finishing parsing `argv`.