mikeakohn/naken_asm #9: ELF machine unknown for several processors

abe-k, 2015-06-25:
The functions for reading and writing ELF files do not generate or recognize the machine number for several of the supported architectures. [The latest version of the gABI spec](http://www.sco.com/developers/gabi/latest/ch4.eheader.html) defines machine numbers for the STM8 (186), the 68k (4), the 68HC08 (71), the 8051 (165), the PowerPC (20), and the Z80 (220) architectures, among others.

mikeakohn, 2015-06-25:
Thanks for letting me know.  I didn't add in PowerPC since there's no real support for it yet, but the others should be there.  I also updated magic_elf to output Z80, STM8, and 8051.  Let me know if you find anything else.

closed 2015-06-25 by mikeakohn

abe-k, 2015-06-25:
Slight typo: you put 168 instead of 186 for the STM8.

mikeakohn, 2015-06-25:
Fixed.