mikeakohn/naken_asm #15: Unary operators not handled

abe-k, 2015-07-21:
The only unary operator present, `~`, for bitwise complement, does not function correctly. As a minimal example,
`.db ~1`
should produce `0xfe` as output but instead produces `0xff`, as it does with any number following the tilde.

To take the complement currently requires treating bitwise complement as a binary operator that ignores its second argument, as in
`.db 1~0`
which produces `0xfe`, with the `0` ignored.

Internally, `OPER_NOT` takes two arguments from the stack and pushes the bitwise complement of the first argument. Since the stack is initialized to contain a zero, giving only the second argument results in an output of the bitwise complement of zero, `0xff`.

At a minimum, `OPER_NOT` should push the bitwise complement of its second argument rather than its first, so that a bare `~1` will function correctly. Much better would be proper handling of unary operators, so they only pop the single argument used.

Also, note that the expression `-1` relies on the initialization of the stack to contain zero, since there is no unary negation operator and the `-` is treated as subtraction. A complete fix would allow `-` to be parsed as either binary (`OPER_MINUS`) or unary (`OPER_NEGATE`, perhaps?) depending on context.

abe-k, 2015-07-26:
The fix introduced in commit 1ffc2f7 only allows the tilde to be immediately followed by a numeric literal. Anything more complicated, such as a parenthesized expression or another unary operator, produces an Unexpected Token error.

It's very common to want a parenthesized expression as the argument of a bitwise complement, such as when creating a bitmask with `~(1<<BIT_POS)`. I can't think offhand of any use cases for composing unary operators (as in `~~1`, `~-1`, or `~+1`), but it would be good to support them for the sake of consistent expression parsing.

mikeakohn, 2015-07-26:
Sorry, I was on vacation when you posted that so I just quickly fixed the simplest case.  I'll look at it now.

abe-k, 2015-07-30:
With the latest commit, taking the bitwise complement of a parenthesized expression now fails with <pre>Error: Expecting ',' but got ')' at <em>filename</em>:<em>lineno</em></pre>.

mikeakohn, 2015-07-30:
I added unit tests for this: tests/unit/eval_expression/unit_test.c and I'm not seeing what you're seeing.  Can you email me some code that fails?

abe-k, 2015-07-30:
I've sent a test file. This is running with a fresh build of commit b86d5dc. I also tried building and running `tests/unit/eval_expression/unit_test.c` and all of those tests pass, even ones that fail when tried from a file.

It seems somewhat that some tokens right after a tilde are ignored, so the assembler doesn't always see the open parenthesis.

abe-k, 2015-07-31:
Seems to be working now, as far as I can tell. Only weird thing remaining that I noticed is that `a~b`, for arbitrary expressions `a` and `b`, is still accepted and is equivalent to `~b`.

closed 2016-05-03 by mikeakohn