mikeakohn/naken_asm #16: Listing file doesn't show includes

abe-k, 2015-07-31:
The contents of files included with `.include "filename.s"` are not shown anywhere in the `.lst` listing file. Any data they declare shows up in `data sections:` at the end, but the source and the assembled machine code are not present in the listing. It would be good if the `.include` directive in the listing file was immediately followed by a listing of the included file's contents (perhaps with some comments marking the beginning and ending), just as if the main file contained all included source. (Alternately, each included file could generate a separate listing file, but this seems less convenient.)

mikeakohn, 2015-09-01:
I was against this idea, but I understand why you'd want it.  The more I think about it though, I could create this as an option.

abe-k, 2015-09-01:
I could see not wanting to incorporate the listing for "standard library"-style includes, but my use case is splitting one program across several files for ease of editing, and I'd like the listing to represent the complete program, not just the relatively small part of it in the main file. (A per-file option would allow both use cases in the same program, but doesn't seem worth the added complexity over a command-line flag.)

hosewiejacke, 2016-04-30:
+1 because of the same use case as abe-k .

mikeakohn, 2017-11-05:
Looks like I added this feature back in July of 2016.  Sorry for not updating this (and forgetting to put the issue number in the check in so it shows up here automatically).

At the top of your include file just put .list and the listing for just that include file will show up in the .lst file that is generated.  I'll go ahead and close the bug.. let me know if this works for you guys.

The feature is also documented in the docs/ directory under "directives".

closed 2017-11-05 by mikeakohn