mikeakohn/naken_asm #20: strange bit in jmp opcode

hosewiejacke, 2016-06-02:
Given this code:

    l: jmp l

Assembling by hand:

* `jmp` gives `3Cxx` (according to slau144j)
* `l ` gives `-1` (jump back 1 word)
* combining `3Cxx` and 9 bits from `-1` gives **`3DFF`**

Assembling with naken_asm gives **`3FFF`** [1]
Who is wrong?

[1]  The full Intel HEX is:

    :02000000ff3dc2
    :00000001FF

mikeakohn, 2016-06-03:
You're talking about MSP430, right?  I think jmp 1 is illegal.  When I try and assemble that with naken_asm I get an error as I would expect.  What's your .org on this also?

hosewiejacke, 2016-06-03:
> You're talking about MSP430, right?

Yes. I forgot naken_asm supports almost any architecture under the sun nowadays.

> I think jmp 1 is illegal.

Sorry for choosing a letter that looks like a digit. I updated the example.

> What's your .org on this also?

There's no .org. I didn't test it on hardware.

mikeakohn, 2016-06-03:
I got almost all of my information from this page:

https://en.wikipedia.org/wiki/TI_MSP430

this page says JMP is 001111 and a 10 bit offset.  jmp to start should be an offset of -2 bytes (-1 words.. or 1111111111)... so I believe 0011111111111111 is correct and the opcode should be 0x3ffff.

hosewiejacke, 2016-06-03:
Thanks. I just looked at the Family User's Guides for

* MSP430x4xx (slau056l, page 3-74)
* MSP430FR4xx and MSP430FR2xx (slau445f, page 172)

and the opcode for jmp is 3Cxx there. I guess it doesn't matter because the 1 bit is a don't-care.

closed 2016-06-03 by hosewiejacke

hosewiejacke, 2016-06-11:
I made a mistake in the calculation above. The jump destination has 10 bits instead of 9.
Sorry.

mikeakohn, 2016-06-11:
I'd much rather getting a bug report like this than getting none at all.  If someone finds an issue (or has an idea how to improve the program) but doesn't bother to let me know, I have a harder time improving the code.  Plus, this also lets me know people have an interest in the assembler which makes me happy.. :)

So thanks for filing this and please let me know if you find anything else...