mikeakohn/naken_asm #26: FR: exit(1) at "Illegal instruction..."

hosewiejacke, 2017-05-01:
When "naken_util -run out.hex" encounters an illegal instruction the exit code is still 0. I think it should be not 0. This would be useful for automated testing.

I hoped to fix it by checking the return value of the simulation, but `simulate_dump_registers_<arch>` (which in turn calls `simulate->context`) does return `void`.

mikeakohn, 2017-05-01:
Good point.  I'll get this fixed by the end of the day.  Thanks for your input!

mikeakohn, 2017-05-01:
I added an extra local variable in main() called error = 0;  If there is a problem with the simulation (and this is auto_run) it flips it to  1 and the naken_util will follow the same code path it always takes (cleaning up properly) but has a return -1 at the end.  Let me know if this works for you and I'll close the issue.  Thanks!

hosewiejacke, 2017-05-02:
Excellent! Works fine. Thank you.

Another idea: `ftell` the position of `stderr` at the beginning and end. If anything gets written to `stderr` the difference will be != 0.

	#include <stdio.h>
	#include <stdlib.h>

	int main() {
		long err0 = ftell(stderr);
		// do something
		long err1 = ftell(stderr);

		// return failure if anything is written to stderr
		long ret = (err1 - err0) == 0 ? EXIT_SUCCESS : EXIT_FAILURE;
		exit(ret);
	}

mikeakohn, 2017-05-03:
This probably sounds odd, but most of my programs typically don't write out to stderr.  But that's an an interesting idea.

closed 2017-05-03 by mikeakohn