mikeakohn/naken_asm #27: calling a non existing label in naken_util seems to call 0

hosewiejacke, 2017-05-12:
Hi Mike,

`call`ing a non existing label in `naken_util` seems to call 0.

## Steps to reproduce:

Generate input:

	$ cat > test.s43
	zero:
		mov #0, r12
		ret

	one:
		mov #1, r12
		ret

<br>

	$ naken_asm test.s43

<br>

	$ naken_util out.hex
	set r12=0xffff
	call nosuchfunc

R12 is zero now. I think there should some kind of warning.

Here's the probably the place where the simulation gets invoked:
https://github.com/mikeakohn/naken_asm/blob/master/simulate/msp430.c#L428

mikeakohn, 2017-05-12:
I'm changing a few things in here.  I'm adding get_address() instead of get_num() that will check the symbol table before it calls get_num().  I'll check in a fix in about 12 hours.

mikeakohn, 2017-05-13:
Should be fixed now.  Thanks for letting me know.

closed 2017-05-13 by mikeakohn

hosewiejacke, 2017-05-15:
Yep, `naken_util` prevents from calling nonexistent symbols now. Thx.
Btw: the prompt after the error message is `running> ` instead of `stopped> `. Is that intended?

mikeakohn, 2017-05-15:
I didn't notice the prompt.  I changed it so it won't switch to running unless the address is okay.  Thanks.