mikeakohn/naken_asm #28: Simulator suggestion: Fail/success status

larsbrinkhoff, 2017-08-11:
Hej!

I'm using your MSP430 simulator to test a compiler.  I compile and run a test program.  If a bad condition is detected, a special routine is called which is supposed to halt and signal failure.  If the test runs to completion, another routine is called to signal success.

Some processors have special ILLEGAL, HALT, or BREAK instructions which I sometimes use for this.  I didn't find any for the MSP430, so I used the illegal opcodes 0000 for success and 0001 for failure which at least makes the simulator halt.

Would it possible to have the simulator detect success/failure for this use case, and exit with a corresponding code?

mikeakohn, 2017-08-11:
Hej hej!  Jag antar du är svensk då?

I think I might not totally understand your use-case.  Are you using the simulator (naken_util) with the -run option so that it runs immediately from the command line?  In this case the program should stop running from an illegal instruction.  It also stops if it gets an "ret" and the stack gets overflowed... or in other words, if your program is running, doesn't use "call" or "push" instructions or anything else that modifies the stack, and then sees an "ret" instruction, the program will stop.

If those options don't help, would it help if naken_util threw an exception and started running code pointed to from an interrupt vector?  Maybe vector 0xfffc?

Let me know what you need and I'll add it in.

larsbrinkhoff, 2017-08-12:
Ja.  Jag såg att din hemsida startade med "Hejsan!".

To be precise, I call naken_util like this:

            printf 'speed 1000\nset pc=0xF000\nrun\nquit\n' | naken_util -msp430 -bin -exe -address 61440 image

I didn't use `-run` because when I did, it started running at address 0.

Yes, the simulator stops when it encounters an illegal address.  What I would like, is for the simulator to exit with a success or failure code.  As in the standard C call `exit(...);`

Your exception idea might work.  I'm open to suggestions.  I just need some way for the target program to communicate success or failure to naken_util, and then for naken_util to return that status with exit().

For example the opcode `0000` could make the simulator do `exit(0);`, and the opcode `0001` could call `exit(1);`.  But it seems a bit of a hack.  Another way might be writing some particular values to an otherwise unused I/O register.

I'm using this to run automated tests in Travis CI.  My repository is http://github.com/larsbrinkhoff/xForth, if you want all the details.  `make check` runs the tests, and the call to naken_util is done from `target/msp430/target.mk`.

Thanks for entertaining my request!  I'm happy to code up your choise of solution and submit a pull request, if you like.

mikeakohn, 2017-08-13:
The MSP430 code gets its start address from the vector in 0xfffe.  So if you did:

.org 0xfffe
  dw start

It should start at whichever address you want.  If you do -run and it hits an illegal instruction, naken_util should return EXIT_FAILURE.

Does that help?  I could add a -start_address switch that could set PC to this value when you use the -run option too.

larsbrinkhoff, 2017-08-13:
Ah, I see.  Starting with -run is slightly different from using the run command.  So executing an illegal instruction or underflowing the stack should provide what I need.  Thanks!

Yes, a -start_address would be helpful.  I suggest it may also be useful to enable auto_run from interactive mode.

mikeakohn, 2017-08-13:
I added -set_pc .. it should be able to take an integer or a hex value.  So you want another option that in interactive mode the program will start automatically?  So you don't have to type "run" from the command line?

larsbrinkhoff, 2017-08-13:
No, that's not it.  What I mean is that it may be nice if the user can do the same things in interactive mode as with command line options.

larsbrinkhoff, 2017-08-14:
I tried using an extra RET instruction to trigger a stack underflow and simulator exit.  However, it doesn't work for my code which sometimes use JMP or BR to exit a subroutine due to tail call optimization.  So the number of CALL instruction doesn't match up with the number of RET instructions.

larsbrinkhoff, 2017-08-14:
I tried to use an extra RET to trigger a stack underflow and simulator exit.  However, this doesn't work with my code due to the fact that it can pop an address from the stack and return using BR.  (This may sound strange, but it's compiled Forth code.)  Thus the CALLs doesn't always match up with the RETs.

mikeakohn, 2017-08-14:
The illegal instruction works though, right?  About the -auto_run.. I think adding that might be confusing with a -run option.  Could the simulator work for you with the -run option (assuming there is a clean way to break out of the running program)?  Hmm.. except you want to be able to adjust the return code based on the result of your running program.

I think here are some options:

1) I use the simulator similar to the way you want to use it for testing Java Grinder.  I always check r15 to get the output of the program?  maybe you can check one of the registers?  This requires no changes in the simulator.

2) Maybe I can add another switch.. like.. -halt_on_addr_0 ... where if you write a value to address 0, the program will exit with the return code you passed to that address?

mikeakohn, 2017-08-15:
The I/O port thing sounds clean to me.  I will add that for you before the end of the day.

mikeakohn, 2017-08-15:
I tested this with:

.msp430

.org 0xc000
main:
  mov.w #1, r5
  mov.w #2, r4
  mov.w #3, r6
  mov.w #1, &0
  jmp main

./naken_util -msp430 -run -set_pc 0xc000 -break_io 0x0000 out.hex

It exits with a return code of 1.  Seems to be working.  Let me know how it goes and if you need anything more.

larsbrinkhoff, 2017-08-16:
That works great, thanks!

closed 2017-08-16 by larsbrinkhoff

larsbrinkhoff, 2017-09-02:
Would you consider something similar for the 6502 simulator too?

mikeakohn, 2017-09-02:
Visst!  Done!  Keep the issues / suggestions coming...

larsbrinkhoff, 2017-09-02:
Works fine, thanks!