mikeakohn/naken_asm #29: Popular targets?

larsbrinkhoff, 2017-08-16:
Since I'm also in the toolchain business, I'm curious.  Do you have a strategy for adding targets?  Do you have some sense which of processors are most popular?

larsbrinkhoff, 2017-08-16:
Here are some for your consideration.

- PDP-11 - nice 16-bit design
- RISC-V - new open source thing
- Xtensa - used in the popular ESP8266 devices
- ColdFire - 68000 recoded

These aren't requests from me, just what popped up when I was looking for processors to support.

mikeakohn, 2017-08-17:
Hmm.. there's RISC-V support now, although I never tested it on a real chip (been resisting getting a HiFive1 board).  I also for some reason thought ColdFire was opcode compatible with 68000 too... I thought just missing some instructions and maybe addressing modes.  Maybe I had it wrong.  I can look into Xtensa.. and PDP-11.. hmmm does anyone use that anymore? :)

Are you wanting just simulators or the assemblers too?  Are you using the chips mentioned above?

larsbrinkhoff, 2017-08-17:
I'm not sure about the ColdFire, I believe I read some time that the instruction encoding was different.  Apparently, some people are still interested in those processors.

PDP-11, just for fun. :-)  It's a nice, regular instruction set.  And a strong influence on successive processors like 68000 and MSP430.

I'm mostly looking for simulators.  When I port to a new processor, I like to be able to run automated tests in e.g Travis CI.

I'm not actually using these devices.  It's mostly for my own amusement.  But I do consider which processors are in current use.

See: http://github.com/larsbrinkhoff/xForth, and  
https://github.com/larsbrinkhoff/xForth/issues/1

mikeakohn, 2017-08-19:
Actually, I have 4 processors on my TODO list right now:  ARC (which is maybe 20% done), THUMB2, webASM, and Java byte code.

I've never done forth before.. your project looks cool though.. I need to clone it and try it out.

larsbrinkhoff, 2017-08-20:
Thanks.  The ARC is a bit of a mystery to me.  It shows up occasionally, but it doesn't seem possible to buy parts.

Feel free to ask if there's anything about my Forth stuff that's hard to understand.

mikeakohn, 2017-08-20:
Arduino 101 has an ARC chip built into it.  But since I haven't finished the assembler yet, I haven't tried to program it.  It's definitely not fun to write an assembler for.

larsbrinkhoff, 2017-08-21:
Thanks for the info!

closed 2017-08-21 by larsbrinkhoff

larsbrinkhoff, 2017-09-01:
About ColdFire, I think you were probably right:

> The ColdFire instruction set is "assembly source" compatible and not entirely object code compatible with the 68000. When compared to classic 68k hardware, the instruction set differs mainly in that it no longer has support for the binary-coded decimal (BCD) packed data format; it removes a number of other, less used instructions; and most instructions that are kept support fewer addressing modes.  The instructions are only 16, 32, or 48 bits long, a simplification compared to the 68000 series.

https://en.wikipedia.org/wiki/NXP_ColdFire#Instruction_set

mikeakohn, 2017-09-01:
Yeah, I think they labeled the instructions in the 68000 manual were labeled as Coldfire compatible.. I thought it was under the CPU32 label but reading up on that it seems that's something else.  It's a shame Motorola dropped this chip in favor of PowerPC... what a neat cheap.