mikeakohn/naken_asm #30: STM8 disassembly strangeness

larsbrinkhoff, 2017-08-30:
Hello,

The STM8 disassembler produces peculiar results when presented with undefined instructions:

    05 9d          ???
    0b 9d          ???
    71 9d          ???
    72 9d 9d       ???
    75 9d          ???
    90 9d 9d       ???
    91 9d 9d       ???
    92 9d 9d       ???

The "71 9D" makes sense I suppose, since the prefix may consume the following opcode.  But I don't see why 05, 0B, and 75 should be two-byte instructions.  Or why the other prefixed instructions are three bytes in length.

mikeakohn, 2017-08-31:
Sorry, didn't get a chance to look into this yet.  Just wanted to make sure you didn't think I was ignoring this.  I should have time tonight.

larsbrinkhoff, 2017-08-31:
Thanks!  It's not urgent.

mikeakohn, 2017-09-02:
That was an odd one.  I had return count + 1 instead of return count if the instruction didn't exist in the STM8 instruction table.  It might have been left over from when I rewrote this assembler.  Let me know if the behavior looks correct so I can close this bug.  Thanks for letting me know about this.

larsbrinkhoff, 2017-09-02:
Thanks, looks good.  I verified with a binary file containing all opcodes padded with NOPs.

#31 looks similar; the instruction byte count is sometimes off by one.

closed 2017-09-02 by larsbrinkhoff