mikeakohn/naken_asm #31: 6502 simulator strange output

larsbrinkhoff, 2017-09-01:
When running in the 6502 simulator, the disassembled output looks strange.  The current instruction looks fine, but subsequent instructions are disassembled at the wrong location, so looks like garbage.

     ! 0x1000: ldx #0xff                                0-0
       0x1002: 0x009a
       0x1003: ldx #0x10                                0-0
       0x1005: 0x004c
       0x1006: bbr3 0x13, 0xfd2 (offset=-54)            0-0
       0x1009: 0x00b5
       0x100b: 0x0095

    Simulation Register Dump                               Stack
    ------------------------------------------------------------
            7 6 5 4 3 2 1 0                          0x1ff: 0x00
    Status: N V - B D I Z C                          0x1fe: 0x00
            1 0 0 0 0 0 0 0                          0x1fd: 0x00
                                                     0x1fc: 0x00
      A=0x00   X=0xff   Y=0x00                       0x1fb: 0x00
     SR=0x80  SP=0xff  PC=0x1003                     0x1fa: 0x00

    0 clock cycles have passed since last reset.

     ! 0x1002: txs                                      0-0
     > 0x1003: ldx #0x10                                0-0
       0x1005: 0x004c
       0x1006: bbr3 0x13, 0xfd2 (offset=-54)            0-0
       0x1009: 0x00b5
       0x100b: 0x0095

    Simulation Register Dump                               Stack
    ------------------------------------------------------------
            7 6 5 4 3 2 1 0                          0x1ff: 0x00
    Status: N V - B D I Z C                          0x1fe: 0x00
            0 0 0 0 0 0 0 0                          0x1fd: 0x00
                                                     0x1fc: 0x00
      A=0x00   X=0x10   Y=0x00                       0x1fb: 0x00
     SR=0x00  SP=0xff  PC=0x1005                     0x1fa: 0x00

    0 clock cycles have passed since last reset.

     ! 0x1003: ldx #0x10                                0-0
       0x1005: 0x004c
       0x1006: bbr3 0x13, 0xfd2 (offset=-54)            0-0
       0x1009: 0x00b5
       0x100b: 0x0095
       0x100c: beq 0x10c3 (offset=-75)                  0-1
       0x100e: 0x00e1

    Simulation Register Dump                               Stack
    ------------------------------------------------------------
            7 6 5 4 3 2 1 0                          0x1ff: 0x00
    Status: N V - B D I Z C                          0x1fe: 0x00
            0 0 0 0 0 0 0 0                          0x1fd: 0x00
                                                     0x1fc: 0x00
      A=0x00   X=0x10   Y=0x00                       0x1fb: 0x00
     SR=0x00  SP=0xff  PC=0x133f                     0x1fa: 0x00

    0 clock cycles have passed since last reset.

     ! 0x1005: jmp 0x133f                               0-0
       0x1008: 0x00ca

larsbrinkhoff, 2017-09-01:
Command line used:

    echo run | naken_util -6502 -set_pc 0x1000 -bin -address 0x1000 image

I can supply the image file if necessary.

mikeakohn, 2017-09-02:
This should be fixed.  Please let me know if this can be closed.

larsbrinkhoff, 2017-09-02:
Looks fine!  Thanks.

closed 2017-09-02 by larsbrinkhoff