mikeakohn/naken_asm #36: MSP430 not using addressing mode optimization

zcsahok, 2017-10-27:
The special case of indirect addressing with zero base should be converted to simple register indirect addressing.
This reduces both code and execution time.

Example: these two operations are equivalent, both should result in one word of code (0x482e).

```
    mov 0(r8),r14

0x0104: 0x481e mov.w 0(r8), r14                         cycles: 3
0x0106: 0x0000

    mov @r8,r14

0x0108: 0x482e mov.w @r8, r14                           cycles: 2

```

mikeakohn, 2017-11-04:
This wasn't as easy of a change as it would seem.  During pass 1 all the addresses of all the labels are calculated.  If the instruction has:  mov.w label(r4), r6 .. and label is defined after it's being used, then the assembler has to assume that "label" isn't a 0 and that it can't optimize.  On pass 2 when label is finally defined, it has to remember that even if label is a 0, it still can't optimize because if it did, all the addresses that follow will change from what they were on pass 1.

It seems like there could be a reason a programmer would not want the optimization to happen.. maybe for timing reasons?  maybe self modifying code?  So I made it so optimizations must be turned on with a command line option -optimize.

mikeakohn, 2017-11-05:
I'll go ahead and close this bug.  Let me know how it goes for you.

closed 2017-11-05 by mikeakohn

zcsahok, 2017-11-05:
Thanks for the quick update. I'm fully with you regarding making this an optional feature.
It does fit my use case well as the offsets are known in advance.

I had the impression that optimization works only if .msp430 is selected explicitly in the file. Not sure if this is intended.

ps: a possible optimization for Z80 could be a generic jump operation that translates to relative or absolute jump depending on the distance of the target.

mikeakohn, 2017-11-05:
I actually had a bug I found last week that if .msp430 wasn't at the top of the file, there was some odd behavior.  That should be fixed now.

For the Z80, would this be .. some kind of "fake" instruction that would get picked based on which would get better code density?