mikeakohn/naken_asm #44: STM8 Instruction "LDF ([$0000.e],X),A" not recognized

fdx1601, 2018-07-05:
Hi, after playing around with the STM8 core recently, I stumbled over an issue with the LDF instruction.

It seems that the assembler truncates addresses below $1000 and then accuses illegal addressing mode.

`
LDF ([$0000.e],X),A    ; does not work
`

mikeakohn, 2018-07-08:
First commit fixes the error and the second makes a check to make sure the address is in range.  Can you check to make sure this fixes the issue?

Thanks for letting me know about this.. and please let me know if anything else doesn't work right.

fdx1601, 2018-07-09:
Hi!

Thanks for the corrections. I wrote a small test routine and it works now as expected.

Test Code:
```
```

And here ist the monitor's output before and after running the Test Code:
```
>d
Enter Lowest  Address: 00:0400
Enter Highest Address: 00:040F

00:0400 FC 81 FF 80 B7 00 DF 00 FE 02 FF 00 FF 00 F7 60

>d
Enter Lowest  Address: 00:0400
Enter Highest Address: 00:040F

00:0400 41 42 43 80 B7 00 DF 00 FE 02 FF 00 FF 00 F7 60
```

Regards!

fdx1601, 2018-07-09:
Forgot to close the issue in my previous post ;-)

closed 2018-07-09 by fdx1601

mikeakohn, 2018-07-09:
Cool.. thanks for the help! :)  Again, let me know if you find anything else...