mikeakohn/naken_asm #54: ARM branches to forward labels cause 'unexpected token' error

fdx1601, 2019-02-26:
Hi!

I  started to play around with the '.arm' assembler and stumbled over this issue, where I get an error every time I use a forward branch to a label.

Example:
```
                    .arm

                    .equ    START           = 0x08000000

                    .org    START

                    BAL     FOO             ; branch forward causes error
                    MOV     R0,R0
FOO:
```

Version: July 1, 2018

Regards.

mikeakohn, 2019-02-26:
I see what the problem is...  I’ll fix it tonight.  Thanks for using the ARM assembler and reporting issues to me.

fdx1601, 2019-02-28:
Thanks a lot for looking into this ... in the meantime I double checked with the newest version and got the same behavior.

Regards.

mikeakohn, 2019-02-28:
Sorry about that. I was tired last night and the fix for this one was a little uglier than I expected.  I just checked in a fix.  Forgot to make the check-in comment for this issue though :(.  But do a git pull and try it again.  It should be working.

Thank you again for testing the ARM assembler and finding this.  This is a huge help to me.

fdx1601, 2019-03-01:
Hi Mike!

Thanks a lot ... your commit r1788 fixed that issue.

But I have run into a similar one for which I'll open another issue.

mikeakohn, 2019-03-01:
Cool.. I'll take care of it :).

closed 2019-03-01 by mikeakohn