mikeakohn/naken_asm #55: ARM load PC relative does not work for labels

fdx1601, 2019-03-01:


fdx1601, 2019-03-01:
Hi!

The last commit got me around the errors, but I'm not sure if the generated opcodes are correct.

When trying to disassemble using naken_util, it does not decode anything for the opcodes generated for `LDR R4,CONST_FOO`. The remaining opcodes are disassembled correctly.

There is another glitch in `naken_util` throwing a `segmentation fault 11` when trying to disassemble a binary file loaded to address `0x8000000`. I'll open another issue on that.

In the meantime I used an online disassembler (https://onlinedisassembler.com/) an tried to disassemble the binary code there.

Disassembly Result for Case 1 (same source as above):

```
.data:08000000 00 00 00 ea                      b	0x08000008
.data:08000004 78 56 34 12                      eorsne	r5, r4, #120, 12	; 0x7800000
.data:08000008 04 40 94 ec                      ldc	0, cr4, [r4], {4}
.data:0800000c fd ff ff ea                      b	0x08000008
```

Disassembly Result for Case 2 (same source as above):

```
.data:00000000 08 40 98 ec                      ldc	0, cr4, [r8], {8}
.data:00000004 fd ff ff ea                      b	0x00000000
.data:00000008 78 56 34 12                      eorsne	r5, r4, #120, 12	; 0x7800000
```

This disassembler interprets the generated opcode as `LDC` instead of `LDR` instruction, whereas the branch opcodes seem OK.

Regards.

mikeakohn, 2019-03-01:
Yeah, it wasn’t meant to be “the fix”.  Again, that problem is a bit bigger than it seems and I’m contemplated a couple different ways to fully take care fo it.  I’ll let you know.

fdx1601, 2019-03-01:
Alright ... no pressure ... take your time ;-)

mikeakohn, 2019-03-03:
This one should be fixed now.  I added to tests/comparison/arm_template.txt and regenerated arm.txt with GNU's assembler (with gen_arm.py) and the results from naken_asm are the same.

fdx1601, 2019-03-03:
Hi!

Thank you very much for looking into this and getting it fixed so lightning fast.

I think we are back on track now ... the assembler seems to produce a consistent result.

It still throws a couple of errors in the first pass, which could be omitted I think.

```
Version: January 26, 2019

 Input file: src/main.asm
Output file: bin/main.bin

Pass 1...
Error: Offset out of range (-4095,4095) at src/main.asm:37
Error: Offset out of range (-4095,4095) at src/main.asm:43
Error: Offset out of range (-4095,4095) at src/main.asm:50
Pass 2...

Program Info:
```

But my main problem is ... **IT DOES NOT BLINK** ... which of course may be my faulty source code ;-)

As of the listing of the disassembler, it seems everything is looking fine. But I don't understand in detail how the relative addressing works to check if the correct values are loaded into the registers. 

The disassembler shows the PC offset in decimal, where the addresses are all in hex.

```
Addr    Opcode    Instruction                              Cycles
------- --------  ----------------------------------       ------
0x0000: e59f4038  ldr r4, [pc, #56]                        ?
0x0004: e5945000  ldr r5, [r4]                             ?
0x0008: e3855602  orr r5, r5, #0x2000 {#0x02, 12}          ?
0x000c: e5845000  str r5, [r4]                             ?
0x0010: e59f4030  ldr r4, [pc, #48]                        ?
0x0014: e5945000  ldr r5, [r4]                             ?
0x0018: e3c55a02  bic r5, r5, #0x200000 {#0x02, 20}        ?
0x001c: e5845000  str r5, [r4]                             ?
0x0020: ea00000c  b 0x58 (48)                              ?
0x0024: e59f401c  ldr r4, [pc, #28]                        ?
0x0028: e5945000  ldr r5, [r4]                             ?
0x002c: e3855a02  orr r5, r5, #0x200000 {#0x02, 20}        ?
0x0030: e5845000  str r5, [r4]                             ?
0x0034: ea000007  b 0x58 (28)                              ?
0x0038: eafffff4  b 0x10 (-48)                             ?
0x003c: 40011000  andmi r1, r1, r0                         ?
0x0040: 40011004  andmi r1, r1, r4                         ?
0x0044: 40011008  andmi r1, r1, r8                         ?
0x0048: 4001100c  andmi r1, r1, r12                        ?
0x004c: 40011010  andmi r1, r1, r0, lsl r0                 ?
0x0050: 40011014  andmi r1, r1, r4, lsl r0                 ?
0x0054: 40011018  andmi r1, r1, r8, lsl r0                 ?
0x0058: e3a04000  mov r4, #0x0 {#0x00, 0}                  ?
0x005c: e2944001  adds r4, r4, #0x1 {#0x01, 0}             ?
0x0060: 0afffffd  beq 0x5c (-12)                           ?
0x0064: e1b0f00e  movs pc, lr                              ?
```

Would it be possible to add the absolute address as a comment after the instruction line?

```
0x0000: e59f4038  ldr r4, [pc, #56] ; 0x00000048           
                                    ^^^^^^^^^^^^
```

This would make it a lot easier to check the addresses.

Regards!

mikeakohn, 2019-03-03:
Do a pull.  The disassembler should output the absolute address.  Which instruction was giving you an offset error on pass 1?

I probably need to make sure every version of those instruction are in the comparison test.  If you want to try them yourself, you can add them to the tests/comparison/arm.txt file and run the gen_python.py script.  Then you can do:  bash run.sh arm  in that directory to make sure naken_asm gets the same result.  You'll have to have the gcc for ARM on your system.  apt install gcc-arm-linux-gnueabi.

Just out of curiosity too.. which chip are you targeting?

mikeakohn, 2019-03-03:
Sorry, noot arm.txt, but arm_template.txt .. running the python script will generate arm.txt

fdx1601, 2019-03-04:
Hi!

As I am still trying to get the blinking example to work, the source is kind of constantly changing trying out different ways and comparing the results.

The chip I am targeting is a STM32F103C8 which is somewhat constraint in terms of flash memory size. A simple blink implemented using the **mbed** framework takes up quite some resources.

```
DATA:    [==        ]  18.0% (used 3680 bytes from 20480 bytes)
PROGRAM: [=====     ]  54.1% (used 35448 bytes from 65536 bytes)
```

Whereas an empty main() compiled using **arm-none-eabi-gcc** generates a binary file bigger than the 64k device's flash memory.

The assembly code I am trying to get to work follows below.

```
;----------------------------------------------------------------------------------------
;
;   MAIN.ASM    Blink the on-board LED
;
;   Device:     STM32F103C8 (blue pill board)
;
;   Assemble:   naken_asm src/main.asm -h -o bin/main.hex
;
;   Upload:     st-flash --format binary --flash=64k write bin/main.bin 0x8000000
;
;----------------------------------------------------------------------------------------

                    .arm

;----------------------------------------------------------------------------------------
; CONSTANT DECLARATIONS
;----------------------------------------------------------------------------------------

                    .equ    GPIOC_CRL       = 0x40011000
                    .equ    GPIOC_CRH       = 0x40011004
                    .equ    GPIOC_IDR       = 0x40011008
                    .equ    GPIOC_ODR       = 0x4001100c
                    .equ    GPIOC_BSRR      = 0x40011010
                    .equ    GPIOC_BRR       = 0x40011014
                    .equ    GPIOC_LCKR      = 0x40011018

                    .equ    START           = 0x08000000

;----------------------------------------------------------------------------------------
; MAIN PROGRAM
;----------------------------------------------------------------------------------------

                    .org    START

SETUP:
                    LDR     R4,L_GPIOC_CRH      ; load R4 with address of GPIOC_CRH
                    LDR     R5,[R4]             ; load content of GPIOC_CRH register
                    ORR     R5,R5,#2<<20        ; set mode 0b10 for output <= 2MHz
                    STR     R5,[R4]             ; store result into GPIOC_CRH register

LOOP:
                    LDR     R4,L_GPIOC_ODR      ; load R4 with address of GPIOC_ODR
                    LDR     R5,[R4]             ; load R5 with content of GPIOC_ODR
                    BIC     R5,R5,#1<<13        ; clear bit #13
                    STR     R5,[R4]             ; store result into GPIOC_ODR register

                    BL      DELAY               ; call DELAY subroutine

                    LDR     R4,L_GPIOC_ODR      ; load R4 with address of GPIOC_ODR
                    LDR     R5,[R4]             ; load R5 with content of GPIOC_ODR
                    ORR     R5,R5,#1<<13        ; set bit #13
                    STR     R5,[R4]             ; store result into GPIOC_ODR register

                    BL      DELAY               ; call DELAY subroutine

                    B       LOOP                ; repeat over and over

;----------------------------------------------------------------------------------------
; LITERAL POOL
;----------------------------------------------------------------------------------------

L_GPIOC_CRL:        .dl     GPIOC_CRL
L_GPIOC_CRH:        .dl     GPIOC_CRH
L_GPIOC_IDR:        .dl     GPIOC_IDR
L_GPIOC_ODR:        .dl     GPIOC_ODR
L_GPIOC_BSRR:       .dl     GPIOC_BSRR
L_GPIOC_BRR:        .dl     GPIOC_BRR
L_GPIOC_LCKR:       .dl     GPIOC_LCKR

;----------------------------------------------------------------------------------------
; SUB ROUTINES
;----------------------------------------------------------------------------------------

DELAY:              .scope
                    MOV     R4,#0               ; load register with zero
LOOP:               ADDS    R4,R4,#1            ; increment by 1
                    BNE     LOOP                ; count until overflow occurres
                    MOVS    PC,LR               ; return to caller
                    .ends
```

Trying to assemble it gives me these pass 1 errors, but nevertheless it produces a complete looking binary at the end. The error lines match the corresponding lines in the source code from above.

```
Version: January 26, 2019

 Input file: src/main.asm
Output file: bin/main.bin

Pass 1...
Error: Offset out of range (-4095,4095) at src/main.asm:36
Error: Offset out of range (-4095,4095) at src/main.asm:42
Error: Offset out of range (-4095,4095) at src/main.asm:49
Pass 2...

Program Info:
```

I will try to follow your suggestions.

Regards.

mikeakohn, 2019-03-04:
Give it a try now...

fdx1601, 2019-03-04:
Hi!

I gave it a try ... but seems we have some kind of regression. The errors in pass 1 are now appearing in pass 2. Additionally the lower range check fell back to use 0 instead of -4095 compared to the previous version.

```
Version: Seg  4 Mar 2019 09:41:30 -03

 Input file: main-behind.asm
Output file: main.bin

Pass 1...
Pass 2...
Error: Offset out of range (0,4095) at main-behind.asm:36
Error: Offset out of range (0,4095) at main-behind.asm:42
Error: Offset out of range (0,4095) at main-behind.asm:49

Program Info:
```

And the disassembler started to show garbage again ...

```
Addr    Opcode    Instruction                              Cycles
------- --------  ----------------------------------       ------
0x8000000: ecd04040                                         ?
0x8000004: e5945000  ldr r5, [r4]                             ?
```

... compared to the previous version:

```
Addr    Opcode    Instruction                              Cycles
------- --------  ----------------------------------       ------
0x0000: e59f4038  ldr r4, [pc, #56]                        ?
0x0004: e5945000  ldr r5, [r4]                             ?
```

BTW I very much appreciate your hard work in getting this fixed!

Regards!

mikeakohn, 2019-03-04:
Odd.. I ran through your code and it worked as expected. On top of that, whenever it gets an error (pass 1 or pass 2) it should be aborting immediately.  You shouldn’t see 3 of those errors in a row.

fdx1601, 2019-03-04:
To make sure we are working with the same codebase I'll trash my local copy obtained via SVN and download a fresh one.

fdx1601, 2019-03-04:
Bingo!

Looking very good ... no errors in any pass ... disassembly looks kind of confident too ;-)

```
Version: January 26, 2019

 Input file: main.asm
Output file: main.hex

Pass 1...
Pass 2...

Program Info:
Include Paths: .
```

```
Addr    Opcode    Instruction                              Cycles
------- --------  ----------------------------------       ------
0x8000000: e59f4038  ldr r4, [pc, #56] ; 0x8000040            ?
0x8000004: e5945000  ldr r5, [r4]                             ?
0x8000008: e3855602  orr r5, r5, #0x2000 {#0x02, 12}          ?
0x800000c: e5845000  str r5, [r4]                             ?
0x8000010: e59f4030  ldr r4, [pc, #48] ; 0x8000048            ?
0x8000014: e5945000  ldr r5, [r4]                             ?
0x8000018: e3c55a02  bic r5, r5, #0x200000 {#0x02, 20}        ?
0x800001c: e5845000  str r5, [r4]                             ?
0x8000020: ea00000c  b 0x8000058 (48)                         ?
0x8000024: e59f401c  ldr r4, [pc, #28] ; 0x8000048            ?
0x8000028: e5945000  ldr r5, [r4]                             ?
0x800002c: e3855a02  orr r5, r5, #0x200000 {#0x02, 20}        ?
0x8000030: e5845000  str r5, [r4]                             ?
0x8000034: ea000007  b 0x8000058 (28)                         ?
0x8000038: eafffff4  b 0x8000010 (-48)                        ?
0x800003c: 40011000  andmi r1, r1, r0                         ?
0x8000040: 40011004  andmi r1, r1, r4                         ?
0x8000044: 40011008  andmi r1, r1, r8                         ?
0x8000048: 4001100c  andmi r1, r1, r12                        ?
0x800004c: 40011010  andmi r1, r1, r0, lsl r0                 ?
0x8000050: 40011014  andmi r1, r1, r4, lsl r0                 ?
0x8000054: 40011018  andmi r1, r1, r8, lsl r0                 ?
0x8000058: e3a04000  mov r4, #0x0 {#0x00, 0}                  ?
0x800005c: e2944001  adds r4, r4, #0x1 {#0x01, 0}             ?
0x8000060: 1afffffd  bne 0x800005c (-12)                      ?
0x8000064: e1b0f00e  movs pc, lr                              ?
```

As soon as I get back home, I'll try to upload the code to see if it works on the real device.

Regards!

fdx1601, 2019-03-05:
Still not blinking but assembly is OK

closed 2019-03-05 by fdx1601

mikeakohn, 2019-03-05:
Is that an ARM Cortex?  Maybe it only does THUMB instructions?

fdx1601, 2019-03-05:
I suspect this is exactly the case ;-)

Luckily I was able to find a very simple example which uses a similar target.

https://jacobmossberg.se/posts/2017/01/10/minimalistic-assembler-program-on-stm32-h103.html

I'll give it a try.

mikeakohn, 2019-03-05:
That’s a bummer.  I think the THUMB code is pretty tight since I had all the comparison tests working on it.  You were really helping me out a lot with the ARM part.  You need to get a chip with an ARM instruction set :).

fdx1601, 2019-03-05:
Yes, indeed I'll have to get one after all those hours spent on this project.

Concerning the THUMB, it seems it doesn't recognize the `ADR` instruction:

Accordingly to the programming manual ...

> **ADR**
>     Load PC-relative address.
> Syntax
>     `ADR{cond} Rd, label`
> where:
> • ‘cond’ is an optional condition code
> • ‘Rd’ is the destination register
> • ‘label’ is a PC-relative expression

mikeakohn, 2019-03-05:
Can you start a new issue for it?  I’ll take care of it tonight.  I havn’t had a chance to write any THUMB code either.. just the testing to make sure instuctions i knew about worked.

fdx1601, 2019-03-05:
done