mikeakohn/naken_asm #56: naken_util disasm throws 'segmentation fault 11' for arm and start address 0x8000000

fdx1601, 2019-03-01:
Hi!

This issue describes a situation where trying to disassemble the following source code loaded at address `0x08000000` generates a `segmentation fault`.

Source code:
```
                    .arm

                    .equ    START           = 0x08000000

                    .org    START

                    B       LOOP

CONST_FOO:          .dl     0x12345678

LOOP:
                    LDR     R4,CONST_FOO

                    B       LOOP
```

Build command:
```
naken_asm -bin foo1.asm -o foo1.bin
```

Trying to dissassemble
```
$ naken_util -arm -bin -address 0x8000000 -set_pc 0x8000000 foo1.bin 

naken_util - by Michael Kohn
                Joe Davisson
    Web: http://www.mikekohn.net/
  Email: mike@mikekohn.net

Version: January 26, 2019

Loaded bin foo1.bin from 0x8000000 to 0x800000f
Type help for a list of commands.
stopped> disasm

Addr    Opcode    Instruction                              Cycles
------- --------  ----------------------------------       ------
Vectors:
Segmentation fault: 11
```

Regards!

mikeakohn, 2019-03-03:
This one should be fixed now.

fdx1601, 2019-03-05:
yep ... this one has gone

closed 2019-03-05 by fdx1601