mikeakohn/naken_asm #57: THUMB doesn't recognize ADR instruction

fdx1601, 2019-03-05:
Hi!

The "ADR" instruction described in the STM32F103C8 programing manual "The Cortex®-M3 instruction Set" is not recognized as a valid instruction when assembling a THUMB source.

> **ADR**
>     Load PC-relative address.
> _Syntax:_
>     `ADR{cond} Rd, label`
> _Where:_
>     • ‘cond’ is an optional condition code (see Conditional execution on page 56)
>     • ‘Rd’ is the destination register
>     • ‘label’ is a PC-relative expression (see PC-relative expressions on page 56)
> _Operation:_
> ADR determines the address by adding an immediate value to the PC. It writes the result to the destination register.
> ADR produces position-independent code, because the address is PC-relative.
> If you use ADR to generate a target address for a BX or BLX instruction, you must ensure that bit[0] of the address you generate is set to1 for correct execution.
> Values of label must be within the range -4095 to 4095 from the address in the PC. You might have to use the .W suffix to get the maximum offset range or to generate
> addresses that are not word-aligned (see Instruction width selection on page 58). Restrictions
> Rd must be neither SP nor PC. Condition flags
> This instruction does not change the flags.
> _Example:_
```
ADR R1, TextMessage ; write address value of a location labelled as
                    ; TextMessage to R1
```

Regards!

fdx1601, 2019-03-06:
Hurray!!!

it's blinking ... it's blnking ... 

Regards!

```
;----------------------------------------------------------------------------------------
;
;   BLINK.ASM   Blink the on-board LED
;
;   Device:     STM32F103C8 (blue pill board)
;
;   Assemble:   naken_asm src/blink.asm -b -o bin/blink.bin
;
;   Debug:      naken_util -arm -bin -address 0x08000000 bin/blink.bin
;
;   Upload:     st-flash --format binary --flash=64k write bin/blink.bin 0x8000000
;
;----------------------------------------------------------------------------------------

                    .thumb

;----------------------------------------------------------------------------------------
; CONSTANT DECLARATIONS
;----------------------------------------------------------------------------------------

                    .equ    START           = 0x08000000
                    .equ    TOP_OF_STACK    = 0x20005000

;----------------------------------------------------------------------------------------
; VECTOR TABLE
;----------------------------------------------------------------------------------------

                    .org    0x08000000

VECTORS:
                    .dl     TOP_OF_STACK        ; the top of stack address
                    .dl     IRQ_RESET+1         ; the reset vector address
                    .dl     IRQ_NMI+1           ; dummy NMI ISR address
                    .dl     IRQ_HARD_FAULT+1    ; Hard Fault ISR address
                    .dl     IRQ_MEMORY_FAULT+1  ; Memory Fault ISR address
                    .dl     IRQ_BUS_FAULT+1     ; Bus Fault ISR address
                    .dl     IRQ_USAGE_FAULT+1   ; Usage Fault ISR address

                    .org    START + 0x2c
                    .dl     IRQ_SERVICE+1       ; System service call via SWI instruction
                    .dl     IRQ_DEBUG+1         ; Debug Monitor

                    .org    START + 0x38
                    .dl     IRQ_PENDING+1       ; pendable request for system service
                    .dl     IRQ_SYSTICK+1       ; system tick timer

                    .org    START + 0x0130      ; address following vector table

;----------------------------------------------------------------------------------------
; INTERRUPT HANDLERS
;----------------------------------------------------------------------------------------

IRQ_RESET:          B       SETUP

IRQ_NMI:
IRQ_SERVICE:
IRQ_DEBUG:
IRQ_PENDING:
IRQ_SYSTICK:
IRQ_HARD_FAULT:
IRQ_MEMORY_FAULT:
IRQ_BUS_FAULT:
IRQ_USAGE_FAULT:
HALT:
                    B       $

;----------------------------------------------------------------------------------------
; MAIN PROGRAM
;----------------------------------------------------------------------------------------

SETUP:
                    LDR     R6,RCC_APB2ENR      ; load address of RCC_APB2ENR register
                    MOV     R0,#0x10            ; enable PORTC clock
                    STR     R0,[R6,#0]          ; store configuration in RCC_APB2ENR

                    LDR     R6,GPIOC_CRH        ; load address of GPIOC_CRH register
                    LDR     R0,GPIOC_RESET      ; load GPIOC_RESET value
                    STR     R0,[R6,#0]          ; configure PORTC pins

                    LDR     R2,TOGGLE_ON        ; load TOGGLE_ON pin states
                    LDR     R3,TOGGLE_OFF       ; load TOGGLE_OFF pin states

                    LDR     R6,GPIOC_ODR        ; load address of GPIOC_ODR register

LOOP:
                    LDR     R4,[R6,#0]          ; load content of GPIOC_ODR register
                    ORR     R4,R2               ; restore TOGGLE_ON pin states
                    STR     R4,[R6,#0]          ; store result in GPIOC_ODR register

                    BL      DELAY               ; call delay subroutine

                    LDR     R4,[R6,#0]          ; load content of GPIOC_ODR register
                    AND     R4,R3               ; restore TOGGLE_OFF pin states
                    STR     R4,[R6,#0]          ; store result in GPIOC_ODR register

                    BL      DELAY               ; call delay subroutine

                    B       LOOP                ; repeat this over and over

;----------------------------------------------------------------------------------------
; SUB ROUTINES
;----------------------------------------------------------------------------------------

DELAY:              .scope
                    LDR     R1,BLINK_DELAY      ; load value of BLINK_DELAY
REPEAT:             SUB     R1,#1               ; subtract 1 from counter
                    CMP     R1,#0               ; check if result is 0
                    BNE     REPEAT              ; keep counting if not 0
                    MOV     PC,LR               ; return to caller
                    .ends

;----------------------------------------------------------------------------------------
; LITERAL POOL
;----------------------------------------------------------------------------------------

                    .align  32

RCC_APB2ENR:        .dl     0x40021018          ; periphal port clock register

GPIOC_CRH:          .dl     0x40011004          ; PORTC configuration register (pins 8..15)
GPIOC_ODR:          .dl     0x4001100C          ; PORTC output register

GPIOC_RESET:        .dl     0x41144444          ; PORTC pin configuration

BLINK_DELAY:        .dl     400000              ; blink delay counter value

TOGGLE_ON:          .dl     (1<<13)|(1<<14)     ; bit mask for ON state
TOGGLE_OFF:         .dl     ~((1<<13)|(1<<14))  ; bit mask for OFF state
```

mikeakohn, 2019-03-06:
Woah!  That's awesome! :)  I added that adr instruction, but something looks off to me.  Maybe I'm just tired but it seems there are the adr instruction conflicts with a version of an add instruction.. I think add r3, SP, #8 for example. I can't seem to get the instruction to assemble with gnu-as also to confirm this and the disassembler disassembles both correctly.

fdx1601, 2019-03-06:
Did you use these directives?

```
    .thumb
    .syntax     unified
```

From what I've seen in related posts the `.syntax unified` is kind of mandatory for gnu-as to assemble correctly.

I've an adopted version of the code above using the `adr` instruction and it assembled correctly and executed fine on the target device.

Regards!

mikeakohn, 2019-03-07:
I had .thumb on there but not .syntax unified.  I added it and the adr does assemble now, but it seems like it's putting out different opcodes for everything now.  I'll have to look into it.

fdx1601, 2019-03-09:
Hi!

I downloaded the latest version and so far it recognizes the `ADR` instruction. But it's aliased to something like `ADD R0, PC, #4`, what seems OK to me, even I thought it was an instruction of its own.

But trying to assemble a somewhat more extense assembly source, I stumbled over some weird behaviour. Trying to assemble the following code snippet will throw an alignment error.

```
                    .thumb

HELP:               .scope
                    PUSH    {LR}
                    ADR     R0,STR_HELP
                    BL      PRINT_STR
                    POP     {PC}

                    .align  32
STR_HELP:           .ascii  "E edit memory\r\n"
                    .ascii  "D dump memory\r\n"
                    .ascii  "\0"
                    .ends
```

even with explicit `.align 32` it results in this error:

```
Error: Offset not 4 byte aligned at src/foo.asm:5
```

Regards

mikeakohn, 2019-03-09:
The alias to "add" doesn't surprise me.  As I was saying earlier, it seemed like it was using the same opcode as another "add" instruction, so I guess that was it.

The alignment thing is tricky.  That instruction takes the PC and adds an offset to it (between 0 and 1020) which is encoded by shifting it to the right by 2 bits.  The offset needs to be a multiple of 4.  In your case, the address of the "adr" instruction is on byte 6.  So since your data section is aligned properly, the offset is off by 2 bytes.  I'm going to add more output to that error so it makes more sense.

fdx1601, 2019-03-09:
Hmmh ... I see ... in fact putting a simple `MOV` instruction before using the `ADR` instruction resolves the issue ...

```
                    MOV     R0,#0               ; used for address padding
                    ADR     R0,STR_HELP
```

I think there isn't much you can do about it ... other than throw some dummy instruction in between.

fdx1601, 2019-03-09:
After using the approach from the previous post it turned out to have to insert and remove the padding instructions every time the source code changes and padding is affected.

Perhaps it could be addressed with some kind of directive ...

```
.if !aligned(32)
                    MOV     R0,#0               ; used for address padding
.endif
                    ADR     R0,STR_HELP
```

fdx1601, 2019-03-10:
Looking again at this I found a note in the programming manual that talks about explicit "Instruction Width Selection" by adding the suffixes `.W` or `.N`.

> In some cases it might be necessary to specify the .W suffix, for example if the operand is the label  of an instruction or literal data, as in the case of branch instructions. This is because the assembler might not automatically generate the right size encoding.

Giving these examples:
```
    BCS.W label         ; creates a 32-bit instruction even for a short branch

    ADDS.W R0, R0, R1   ; creates a 32-bit instruction even though the same
                        ; operation can be done by a 16-bit instruction
```

The documentation for `ADR` refers to this section too.

Perhaps a way to getting around it.

Regards.

mikeakohn, 2019-03-10:
I could have it automatically insert a NOP before the instruction... no .w is needed.

fdx1601, 2019-03-10:
ok, that sounds great!

mikeakohn, 2019-03-10:
Actually, I can't do that.  Because if that's done then the instruction size will either be 2 bytes or 4 bytes and the assembler won't know until pass 2.  On pass 1 it assumes 2 bytes and all the addresses after that are calculated based on that.  If it changes to 4 on pass 2, all the addresses will be wrong.

fdx1601, 2019-03-10:
How about generating always a 4 byte instruction?

Pass 1 always with following `NOP`:
```
ADR    R0,FOO
NOP
```

Pass 2 adjust swapping the instructions if required:
```
NOP
ADR    R0,FOO
```

Just a thought.

mikeakohn, 2019-03-10:
Feels like it's punishing people who do align that instruction.  If it gets a misaligned error, the user can drop their own nop in there when they notice it.

mikeakohn, 2019-03-10:
I have a different idea.. although I don't know how to write the syntax without it being ugly.  Maybe there could be a special align directive that aligns offsets.  Like:

instr_here:
  ldr r4, data

.offset_align 32, instr_here
data:
  db "here's some data"

Although I kind of don't like bit aligns.. I prefer 4 instead of 32, but that doesn't match .align then.

fdx1601, 2019-03-10:
Mmmh ... seems that won't work out well when loading from different locations, like:

```
A:      ldr r0,data
...
B:      ldr r0,data
...
.offset_align 32, ???
data:
        db "here's some data"
```

Would it be possible do it this way?
```
        .pad data          ; insert NOP as required with respect to 'data' offset
        ldr r0,data
        ...
        .pad data          ; insert NOP as required with respect to 'data' offset
        ldr r0,data

        .align 32
data:
        db "here's some data"
```

fdx1601, 2019-03-10:
BTW I agree on preferring byte over bit alignment

mikeakohn, 2019-03-10:
Good point.  The .pad data still won't work because on pass 1 the assembler won't know if it should pad or not.  Pass 1 is there only to make calculations on what addresses the symbols resolve to.

fdx1601, 2019-03-10:
Forgot to mention that I was thinking the `.pad` could do the NOP swapping thing as of my earlier post.

I understood your concern about not wanting to penalty users who do their own padding. 

The explicit `.pad` instruction could take care of this, leaving code where it isn't used as is.

If it could be written in one line like:
```
.pad    adr r0,FOO
```
This would be kind of convenient ... but perhaps this violates some other aspect, even the code doesn't look ugly.

fdx1601, 2019-03-16:
I arranged me with this by now. And the Cortex-M3 programming reference also mentions that the NOP may have to be used to address this kind of problem:

> Use NOP for padding, for example to place the following instruction on a 64-bit boundary.

Let's close that too.

closed 2019-03-16 by fdx1601