mikeakohn/naken_asm #58: Feature idea: dumpram while simulation running

NullMember, 2019-03-09:
Hi
I'm writing new 6502 program and i need to see RAM changes while program running. Stopping and printing RAM is not efficient. While playing with naken_util I see dumpram option but it is not implemented in 6502 simulation. I've imlemented it but it is not working like what I expected. I thought it's dumping RAM during simulation running. So I wrote some changes and now I'm calling simulate_dumpram function in simulate_run. Here is what it looks like

[image]

I can make PR if anyone want to see how I implemented it. I will like to see suggestions of implementation also.

mikeakohn, 2019-03-10:
So you want a constant dump of RAM while the simulator is running?  The 6502 can address 64k of RAM, how would the simulator know what addresses to constantly dump?

NullMember, 2019-03-10:
User need to be write "dumpram startaddress-endaddress". In this example I wrote "dumpram 0x00-0x20" before running simulator to dump between these addresses

closed 2019-08-15 by NullMember