mikeakohn/naken_asm #60: THUMB missing MSR and MRS instructions

fdx1601, 2019-03-09:
Hi!

I wanted to set the carry flag and accordingly to the programing manual this is done by using the `MSR` and `MRS` instructions, which the assembler fails to recognize.

```
        MSR R0,APSR        ; copy APSR (application program status register) into R2
        ...
        MRS APSR,R0        ; copy R2 back into APSR
```

Regards.

mikeakohn, 2019-03-09:
Those two instructions are soooooo ugly.  The documentation shows them as 32 bit with a bunch of binary constants surrounded by ( ).. like.. (0) (1).  I think that's why I didn't implement them.. because I didn't know what to put there.  I'll see if I can get gnu-as to give output and just copy what they did.

fdx1601, 2019-03-09:
Yeah it's quite a bit cumbersome to set/clear a bit in the status register. Perhaps they should have specified a dedicated instruction for this.

Anyway ... for the time being, I am using the -1 constant to indicate an error condition.

But this is also takes a couple of instructions :-(

```
MOV R2,#1
CMN R1,R2
BEQ ERROR
```

fdx1601, 2019-03-10:
Looking into the commit seems that this has been a tough one ;-)

mikeakohn, 2019-03-10:
Yeah.. 32 bit wide instructions sitting in the middle of a 99.9% 16 bit wide instruction set makes things tough.  But then again.. ARM and THUMB are both a bit ugly.. compared to some cleaner instruction sets. Probably part of the reason I haven't done a project with it yet.. :)

fdx1601, 2019-03-10:
Ooops ... Houston we got a problem ...

when assembling my source with the most recent version I get this:

```
Pass 1...
Error: Unknown operands combo for 'BMI' at src/main.asm:257.
```

and may `src/main.asm:257` looks like this:
```
                    CMP     R1,#32
                    BMI     CONTROL
                    ...
CONTROL:
                    ...
```

The previous version assembled without throwing any error.

fdx1601, 2019-03-10:
As of STM32F103C8 programming manual there were `basepri` and `faultmask` registers missing in 
`special_reg_thumb`:

```
struct _special_reg_thumb special_reg_thumb[] =
{
  { "apsr",       0 },
  { "iapsr",      1 },
  { "eapsr",      2 },
  { "xpsr",       3 },
  { "ipsr",       5 },
  { "epsr",       6 },
  { "iepsr",      7 },
  { "msp",        8 },
  { "psp",        9 },
  { "primask",   16 },
  { "basepri",   17 },
  { "faultmask", 19 },
  { "control",   20 },
  { NULL,         0 }
};
```

BTW The register numbers have been taken from corresponding gnu-as disassembled object code.

mikeakohn, 2019-03-10:
I wonder if those are specific to that CPU.  If you like you can submit merge requests for those.

fdx1601, 2019-03-10:
I think there is nothing CPU specific in the gnu-as command line and it recognized them as valid. 
```
arm-none-eabi-as -mcpu=cortex-m3 -mthumb -mapcs-32 -gstabs src/foo.s
```
Suppose they are common to the M3 architecture.

fdx1601, 2019-03-10:
BTW the gnu-as didn't like the `APSR` register and doesn't assemble instructions trying to access them.

On the other hand, when accessing the `PSR` register with the `MSR PSR,R0` it complaints:

```
writing to APSR without specifying a bitmask is deprecated
           ^^^^
```

After some research I found someone has to write it like:
```
MSR     PSR_nzcvq,R0
```
The disassembly generated by gnu-objdump shows both lines identical as:
```
f380 8803 	msr	PSR, r0
```

hosewiejacke, 2019-03-11:
> But then again.. ARM and THUMB are both a bit ugly.. compared to some cleaner instruction sets.

Just curious: what ISAs do you consider being clean?
Because it seems you're familiar with quite a few of them ;)

	$ ls asm/*.c | grep -v common | grep -v template | wc -l
	32

mikeakohn, 2019-03-11:
Well, it’s just my opinion of course.. but the cleanest to me are msp430, MIPS, and avr8.  From having to write an assembler, MSP430 (without the 24 bit instructions) is a simple instruction set.  MIPS too. All their instructions fit a very simple pattern.  From a programmer’s point of view those 3 instruction sets make the most sense.  Easy to memorize the syntax and instruction names. The negatives of those msp40 / MIPS being the syntax when doing indexing.  Using ( ) around the indexing register instead of [ ] like Intel makes writing an assembler much more difficult.

fdx1601, 2019-03-11:
The MSP430 is definitely one of the more consistent instructions sets. I believe when we use "elegant" and "ugly" it is tightly related with consistency.

In design there is a statement saying "form follows function" ... we could adopt this by saying "instruction set follows architecture".

Most of the "weird" things have it's reason in the underlying architecture.

Speaking for the ARM/THUMB instruction set is has some nice concepts:
- push/pop list of registers
- literal pooling

If you think about "literal pooling", which I found very odd initially, it saves 4 bytes of code space over traditional absolute addressing. When used only once, it doesn't make a difference over absolute addressing, as you would use 6 bytes for a traditional load/save operation, 2 for the opcode and 4 for the address. But when loading multiple times (given it can't remain in a dedicated register) you save 4 bytes on every usage.

Examples:
```
MOV R0,#0x12345678     ; not possible because won't fit in a 16-bit instruction word ... neither in a 32-bit one
```

```
LDR R0,MY_LITERAL      ; loading 32-bit value into R0 using 16-bit instruction word
...
MY_LITERAL .dl 0x12345678
```

Regards.

fdx1601, 2019-03-11:
BTW last time I tried to compile my code it still threw this error:

```
Pass 1...
Error: Unknown operands combo for 'BMI' at src/main.asm:257.
```

Do you prefer me to open a separate issue on that?

mikeakohn, 2019-03-12:
Please open a separate issue (and close this one if you feel it's fixed).  And if you could, give me the smallest program possible that  cause it (3 lines of code would be good).  I tried a piece of code with bmi and it appeared to work.

fdx1601, 2019-03-12:
Instruction wise it seems to be OK. 

Haven't checked the result on the real device, since source doesn't assemble due to other issue.

closed 2019-03-12 by fdx1601