mikeakohn/naken_asm #62: THUMB fails to recognize MOVS and SUBS instructions among others

fdx1601, 2019-03-09:
Hi!

the current version of the THUMB assembler doesn't accept 'S' flavors of some instructions. 

One critical usage, is the `MOVS` instruction to return from an interrupt routine. Maybe there is a workaround though which I weren't able to find out.
```
                    MOVS    PC,LR
```

Cross-checking with documentation I figured out that none of these variants are accepted:
```
                    MOVS    PC,LR
                    MVNS    R1,R2

                    ADCS    R1,R2
                    ADDS    R1,R2
                    SBCS    R1,R2
                    SUBS    R1,R2
                    ANDS    R1,R2
                    BICS    R1,R2
                    EORS    R1,R2
                    MULS    R1,R2
                    ORNS    R1,R2
                    ORRS    R1,R2

                    ADCS    R1,R2,R3
                    ADDS    R1,R2,R3
                    SBCS    R1,R2,R3
                    SUBS    R1,R2,R3
                    ANDS    R1,R2,R3
                    BICS    R1,R2,R3
                    EORS    R1,R2,R3
                    MULS    R1,R2,R3
                    ORNS    R1,R2,R3
                    ORRS    R1,R2,R3
```

mikeakohn, 2019-03-10:
I just noticed this also.  naken_asm takes mov r12, r14 and in the pdf I'm looking at it shows up at movs <rd>, <rm>.  The comparison test puts mov r12, r14 into gnus-as and it assembles.  I put movs r12, r14 in there and it assembles the same way.

I wonder if I should just change all the instructions so it forces the user to put "S" at the end?  I don't think there is any difference in the instruction encoding?

fdx1601, 2019-03-10:
Indeed, when assembling with gnu-as it always generates the `MOVS` opcodes. But I am not sure if this is correct either.

As of the documentation I am looking at it says:
> ‘S’ is an optional suffix. If S is specified, the condition code flags are updated on the result of the operation

Strange there isn't a different opcode then.

Looking at [this](http://www.coranac.com/files/gba/re-ejected-thumbref2.pdf) opcode reference there are none of the `S` variants listed.

Perhaps it's better stay with that and just document it the way it is.

mikeakohn, 2019-03-16:
Is this bug closable?

fdx1601, 2019-03-16:
Yep, nothing to be done, let's close it.

closed 2019-03-16 by fdx1601