mikeakohn/naken_asm #67: THUMB fails to accept escaped single quote character in immediate operand

fdx1601, 2019-03-12:
previous version assembles this without complain:
```
                    MOV     R1,#'\''
```

new version generates error in pass 1:
```
Pass 1...
Error: Unterminated quote at src/main.asm:309
```

maybe this is not related explicitly to the THUMB assembler.

fdx1601, 2019-03-12:
changing the code in `thumb.c` starting at line 336 from:
```
        if (ch == '\\')
        {
          ch = tokens_get_char(asm_context);
          if (ch == 'n') { ch = '\n'; }
          else if (ch == 'r') { ch = '\r'; }
          else if (ch == 't') { ch = '\t'; }
          else { tokens_unget_char(asm_context, ch); ch = '\\'; }
        }
```

to this adding cases for single and double quotes:
```
        if (ch == '\\')
        {
          ch = tokens_get_char(asm_context);
          if (ch == 'n') { ch = '\n'; }
          else if (ch == 'r') { ch = '\r'; }
          else if (ch == 't') { ch = '\t'; }
          else if (ch == '\'') { ch = '\''; }
          else if (ch == '\"') { ch = '\"'; }
          else { tokens_unget_char(asm_context, ch); ch = '\\'; }
        }
```
resolves this issue.

mikeakohn, 2019-03-14:
Just updating you so you don’t think I forgot about this.  I wrote some unit tests to catch this stuff .. and I have kind of a different fix for what’s going on here.  Hopefully I have it completed tonight.

fdx1601, 2019-03-14:
I am fine with that ... at the moment everything is working great.

As part of my tiny serial monitor I'll give me the challenge to dive deeper into interrupt handling. 

My goal is to find a way to intercept hardware exception events when reading from or writing to invalid memory addresses, so I can avoid the device to freeze like it does now.

Happy coding ;-)

mikeakohn, 2019-03-16:
This should be good now.. and protected with tests.

fdx1601, 2019-03-16:
OK, issue fixed.

closed 2019-03-16 by fdx1601