mikeakohn/naken_asm #72: THUMB Feature Request: .az32 directive

fdx1601, 2019-03-16:
When writing assembly for THUMB certain instructions require 4-byte aligned addresses or PC offsets. Where this is generally not a problem for data words, it becomes a problem when defining arbitrary long data bytes like strings for example.

To get around this, one has to use explicit alignment:
```
                    .align  32
STR1:               .asciiz "um"
                    .align  32
STR2:               .asciiz "dois"
                    .align  32
STR3:               .asciiz "tres"
                    .align  32
STR4:               .asciiz "quatro"
```
It would be nice to have a directive which combines these two in one:
```
STR1:               .az32   "um"
STR2:               .az32   "dois"
STR3:               .az32   "tres"
STR4:               .az32   "quatro"
```

Happy coding ;-)

mikeakohn, 2019-03-17:
I'm not a huge fan of adding new directives... my goal was to keep the assembler as simple as possible so it's easy to learn.  If you wanted to you could make a macro:

.macro STRING(label, data)
  .align 32
label:
  .ascii data
.endm

STRING(string_1, "TEXT")
STRING(string_2, "BLAH1")
STRING(string_3, "RR")

fdx1601, 2019-03-18:
Ok, a macro will do the trick. Thanks for the hint.

closed 2019-03-18 by fdx1601