mikeakohn/naken_asm #81: test broken with cpu 1802 and instruction "adci 14"

benoitOpenBSD, 2020-04-22:
Hi,

With "make tests", I have the following error ; 

...
[1802]testing req ... PASS
[1802]testing seq ... PASS
[1802]testing adci 14 ... head: 1802_naken.hex: No such file or directory
FAIL
  (exp) :020000007C0E74
  (got) 

And directly : 

naken_asm

Authors: Michael Kohn
         Joe Davisson
    CPU: 1802, 4004, 6502, 65C816, 68HC08, 6809, 68000, 8048, 8051, 86000,
         ARM, AVR8, Cell BE, Copper, CP1610, dsPIC, Epiphany, Java, LC-3,
         MIPS, MSP430, PIC14, PIC24, PIC32, Playstation 2 EE, PowerPC,
         Propeller, PSoC M8C, RISC-V, SH-4, STM8, SuperFX, SWEET16,
         THUMB, TMS1000, TMS1100, TMS9900, WebAssembly, Xtensa, Z80
    Web: http://www.mikekohn.net/
  Email: mike@mikekohn.net
Version: March 15, 2020

 Input file: 1802.asm
Output file: out.hex

Pass 1...
Error: Immediate out of range (-128,255) at 1802.asm:4
** Errors... bailing out

Program Info:
Include Paths: .
               /usr/local/share/naken_asm/include
               include
 Instructions: 0
   Code Bytes: 0
   Data Bytes: 0
  Low Address: fffffffe (4294967294)
 High Address: 0000 (0)

*** Failed ***

With the source code : 

.asm                                                                                                                               <
.1802

start:
  adci 14

Best regards,
Benoit

mikeakohn, 2020-04-22:
Freaky.  Can't reproduce this one (yet).  Based on your username I thought to try it on OpenBSD but it appears it won't compile there .. I think because of missing bash and gmake.  I'll look into it.

benoitOpenBSD, 2020-04-22:
Same result with Debian 10 x86_64 :  
...
[1802]testing req ... PASS
[1802]testing seq ... PASS
[1802]testing adci 14 ... head: cannot open '1802_naken.hex' for reading: No such file or directory
FAIL
  (exp) :020000007C0E74
  (got) 
make[1]: *** [Makefile:4: default] Error 1
make[1]: Leaving directory '/tmp/naken_asm/tests/comparison'
make: *** [Makefile:74: tests] Error 2

mikeakohn, 2020-04-22:
Finally got it to fail.. on MacOS.

mikeakohn, 2020-04-22:
I had an uninitialized variable in 1802.c and xtensa.c.  I guess my system was setting them to the random number of 0.  If it works for you, feel free to close it.  Thanks for letting me know. :)

benoitOpenBSD, 2020-04-23:
Running fine on OpenBSD and Debian Linux 10, thanks !

closed 2020-04-23 by benoitOpenBSD

mikeakohn, 2020-04-23:
Awesome!  Please let me know if you find anything else.