mikeakohn/naken_asm #82: ds

iddq, 2020-06-02:
hello,

can you please show an example how could I fill the memory with 0xff between end of program code and vector table?

I don't find ds in the documentation
where could I replace zero to 0xff in source code if needed?

I tried the following with no success:

```
curr:
ds vec-curr

org 0xfffe

vec:

dw 0xe000

```

thank you.

mikeakohn, 2020-06-04:
That probably needs to end with an error.  The assembler is 2 pass, the first pass is for calculating the addresses of all the labels.  With the code you have above, the address of vec isn't known until pass 2 so it doesn't know how many bytes to allocate.

iddq, 2020-06-08:
what is the solution to initialize the unused bytes with 0xff?

mikeakohn, 2020-06-08:
There is no solution for it now... I'm guessing you're wanting me to add it?  Is this a feature of other assemblers and if so what is the syntax?

iddq, 2020-06-09:
there are several options: I could modify my copy of source code to initialize the memory with 0xff but I did not find yet where it is initialized. or you could implement a directive like dup, times or ds (if i remember correctly) ... and I've found parse_ds function in your source code but I did not find it in the documentation...

mikeakohn, 2020-06-09:
I'm not sure why I added "ds", but I wanted to take it out.  I think you maintaining your own version sounds unfun for you.

I'll add something in the next few days.

mikeakohn, 2020-06-16:
did adding ".data_fill" help?

iddq, 2020-06-16:
yes, thank you

closed 2020-06-16 by iddq