mikeakohn/naken_asm #83: Issues with TMS-1000/TMS-1100 disassembly and possibly assembler

ratsputin, 2020-06-08:
I'm seeing an issue with running the disassembler against some TMS-1100 code (from a TMS-1400 processor).

The command I'm using is: `c:naken_util.exe -disasm -tms1100 darktower.bin`

I'm seeing a couple of issues, one with the disassembly itself which I'll detail momentarily, but the bigger problem seems to be with addressing.  If you look at the section of code pasted below, you'll see that everything seems to work fine in page 0; however, when the disassembler switches pages to page 1 it seems to lose its way:

```
03a 0/0/15: 67     tcmiy 14                                 6
03b 0/0/2a: 2c     ldx 0                                    6
03c 0/0/14: 40     tcy 0                                    6
03d 0/0/28: 6f     tcmiy 15                                 6
03e 0/0/10: bd     br 0x3d (linear_address=0x08)            6
03f 0/0/20: 00     mnea                                     6
040 0/1/00: 28     ldx 0                                    6
041 0/1/01: 28     ldx 0                                    6
042 0/1/03: 28     ldx 0                                    6
043 0/1/07: 28     ldx 0                                    6
044 0/1/0f: 21     tma                                      6
045 0/1/1f: 28     ldx 0                                    6
046 0/1/3f: 28     ldx 0                                    6
```

Here is the same code from the python disassembler I'm currently using:

```
	tcmiy #14        ; $015: $67        // 
	ldx   #1         ; $02a: $2c        // 
.L014
	tcy   #0         ; $014: $40        // 
	tcmiy #15        ; $028: $6f        // 
	br    L03d       ; $010: $bd        // 
	mnea             ; $020: $00        // 

; *******************************************************

	page 1
.L040
	tcy   #8         ; $040: $41        // 
	setr             ; $041: $0d        // 
	tcy   #9         ; $043: $49        // 
	setr             ; $047: $0d        // 
	ldx   #6         ; $04f: $2b        // 
.L05f
	tcy   #0         ; $05f: $40        // 
```
I don't know if this is a side-effect of the code being 4K in size and something is overflowing or what, but the rest of the disassembly is incorrect and looks mostly like the example above.

The other issue deals with the decoding of the `ldx` opcode.  It appears the constant is being decoded incorrectly.  Per the documentation for the operation:
```
 -------------------------------
| 0 | 0 | 1 | 0 | 1 |   | F |   |
 -------------------------------
                     LSB     MSB
```
You'll see from the example I pasted, a `0x2c` is being interpreted as `ldx 0`.  The low-order nybble is a `0xc` which is `0b1100` which should be interpreted as `ldx 1`.

Looking at a bit of your code, it appears you already deal with this bit-swapping in 4-bit constants; the `ldx` operation is the only one that uses a 3-bit constants--it seems to have been designed to accommodate the 3-bit RAM file identifier and is unique to the TMS-1100.

If you're interested in testing with the file I'm using, it can be downloaded from [here](http://www.seanriddle.com/darktower.bin).

ratsputin, 2020-06-08:
I should mention, I just noticed the `-bin` switch.  I tried running with that and it didn't make a difference.

ratsputin, 2020-06-08:
One additional comment that may be related to the addressing issue.  If I'm interpreting the address representation correctly (chapter/page/address), it appears the calculations are incorrect.  Pages are numbered from 0-3, then the chapter increments.  Chapter never exceeds 1.

Per the TMS-1400 docs, a page is 64 eight-bit words and a chapter is made up of 16 pages.  In the case of the TMS-1400 with 4K of ROM, that would be four chapters total.

ratsputin, 2020-06-08:
Okay, I was able to track down and resolve the issues I've identified in the disassembler.  Unfortunately, my abilities with github are pretty much limited to being able to do git clones and open tickets as my C programming experience predates github by more than a decade.

The fixes are as follows in `disasm/tms1000.c`:

The following fix addresses an overflow in references to `tms1000_address_to_lsfr[]` which only has 64 elements:
**Line 19**:
```
define READ_RAM(a) memory_read_m(memory, (tms1000_address_to_lsfr[a & 0x3f]) + (a & 0xfc0))
```

The following fix addresses the `ldx <constant>` issue where the constant is only three bits; the `tms1000_reverse_constant[]` table is based on four-bit values:
**Line 133**:
```
c = tms1000_reverse_constant[(opcode & 0x7) << 1];
```

The following fixes address incorrect calculations on chapter and page.  The page identifier is four bits wide; the chapter identifier is two bits wide:
**Lines 201 and 202**:
```
  uint16_t page = (start >> 6) & 0xf;
  uint16_t chapter = (start >> 10) & 0x3;
```
**Lines 279 and 280**:
```
    uint8_t chapter = (start >> 10) & 0x3;
    uint8_t page = (start >> 6) & 0xf;
```

I certainly don't claim that these are the most elegant solutions to the issues; they do, however, appear to work.  I haven't yet looked at the assembler nor the simulator to see if they share similar issues.

ratsputin, 2020-06-08:
Here is the same chunk of code after the fixes:
```
03a 0/0/15: 67     tcmiy 14                                 6
03b 0/0/2a: 2c     ldx 1                                    6
03c 0/0/14: 40     tcy 0                                    6
03d 0/0/28: 6f     tcmiy 15                                 6
03e 0/0/10: bd     br 0x3d (linear_address=0x08)            6
03f 0/0/20: 00     mnea                                     6
040 0/1/00: 41     tcy 8                                    6
041 0/1/01: 0d     setr                                     6
042 0/1/03: 49     tcy 9                                    6
043 0/1/07: 0d     setr                                     6
044 0/1/0f: 2b     ldx 6                                    6
045 0/1/1f: 40     tcy 0                                    6
046 0/1/3f: 60     tcmiy 0                                  6
```
Here's a chunk of code from the end of that ROM dump showing the corrected Chapter/Page/Addr values:
```
fef 3/f/11: 89     br 0x09 (linear_address=0x32)            6
ff0 3/f/22: 10     ldp 0                                    6
ff1 3/f/04: 0b     clo                                      6
ff2 3/f/09: 12     ldp 4                                    6
ff3 3/f/13: bf     br 0x3f (linear_address=0x06)            6
ff4 3/f/26: 10     ldp 0                                    6
ff5 3/f/0c: 0b     clo                                      6
ff6 3/f/19: 12     ldp 4                                    6
ff7 3/f/32: 80     br 0x00 (linear_address=0x00)            6
ff8 3/f/25: 18     ldp 1                                    6
ff9 3/f/0a: 0b     clo                                      6
ffa 3/f/15: 13     ldp 12                                   6
ffb 3/f/2a: a7     br 0x27 (linear_address=0x10)            6
ffc 3/f/14: 6a     tcmiy 5                                  6
ffd 3/f/28: 25     tamiyc                                   6
ffe 3/f/10: 80     br 0x00 (linear_address=0x00)            6
fff 3/f/20: 94     br 0x14 (linear_address=0x3c)            6
```

ratsputin, 2020-06-08:
Okay, I've shifted to working on the assembly side of things but I'm using my python disassembler as its output can be more easily consumed by `naken_asm` than the output of `naken_util`.

Here is a section of the input code:
```
.tms1100

; *******************************************************
; chapter 0 page 0

L000:
	ldx   0          ; $000: $28       
	tma              ; $001: $21       
	a9aac            ; $003: $71       
	br    L03e       ; $007: $be       
L00f:
	tcmiy 8          ; $00f: $61       
	tcy   0          ; $01f: $40       
	br    L01b       ; $03f: $9b       
L03e:
	tcmiy 0          ; $03e: $60       
L03d:
	ldx   0          ; $03d: $28       
	tcy   0          ; $03b: $40       
	tma              ; $037: $21       
	ldx   4          ; $02f: $29       
	mnea             ; $01e: $00       
	br    L033       ; $03c: $b3       
	retn             ; $039: $0f       
```
Note, each line includes the address and the original value from the bin file.  This is, of course, just the first few lines of the asm file.

Here is the assembler's output:
```
.tms1100

; *******************************************************
; chapter 0 page 0

L000:
        ldx   0          ; $000: $28

000 0/0/00: 28 ldx 0                                    cycles: 6

        tma              ; $001: $21

001 0/0/01: 21 tma                                      cycles: 6

        a9aac            ; $003: $71

002 0/0/03: 71 br 0x07 (linear_address=0x03)            cycles: 6

        br    L03e       ; $007: $be

003 0/0/07: be tcmiy 0                                  cycles: 6

L00f:
        tcmiy 8          ; $00f: $61

004 0/0/0f: 61 ldp 1                                    cycles: 6

        tcy   0          ; $01f: $40

005 0/0/1f: 40 br 0x00 (linear_address=0x00)            cycles: 6

        br    L01b       ; $03f: $9b

006 0/0/3f: 9b mnea                                     cycles: 6

L03e:
        tcmiy 0          ; $03e: $60

007 0/0/3e: 60 br 0x3d (linear_address=0x08)            cycles: 6

L03d:
        ldx   0          ; $03d: $28

008 0/0/3d: 28 tcmiy 15                                 cycles: 6

        tcy   0          ; $03b: $40

009 0/0/3b: 40 ldx 2                                    cycles: 6

        tma              ; $037: $21

00a 0/0/37: 21 tam                                      cycles: 6

        ldx   4          ; $02f: $29

00b 0/0/2f: 28 ldx 4                                    cycles: 6

        mnea             ; $01e: $00

00c 0/0/1e: 00 br 0x17 (linear_address=0x1e)            cycles: 6

        br    L033       ; $03c: $b3

00d 0/0/3c: b3 tcy 0                                    cycles: 6

        retn             ; $039: $0f

00e 0/0/39: 0f ldx 4                                    cycles: 6
```
I'm a bit confused as, with the exception of the `ldx` opcode (as expected), the assembler is generating the right values but the wrong disassembly.

The other issue here is that I'm having to assemble a cut down version of the binary with a single chapter.  If I try to build the entire file, I get the following:
```
naken_asm

Authors: Michael Kohn
         Joe Davisson
    CPU: 1802, 4004, 6502, 65C816, 68HC08, 6809, 68000, 8048, 8051, 86000,
         ARM, AVR8, Cell BE, Copper, CP1610, dsPIC, Epiphany, Java, LC-3,
         MIPS, MSP430, PIC14, PIC24, PIC32, Playstation 2 EE, PowerPC,
         Propeller, PSoC M8C, RISC-V, SH-4, STM8, SuperFX, SWEET16,
         THUMB, TMS1000, TMS1100, TMS9900, WebAssembly, Xtensa, Z80
    Web: http://www.mikekohn.net/
  Email: mike@mikekohn.net
Version: April 25, 2020

 Input file: darktower.asm
Output file: out.hex
  List file: out.lst

Pass 1...
Error: Page out of range (0,15) at darktower.asm:1371
** Errors... bailing out

Program Info:
Include Paths: .
               /usr/local/share/naken_asm/include
               include
 Instructions: 1024
   Code Bytes: 1024
   Data Bytes: 0
  Low Address: 0000 (0)
 High Address: 03ff (1023)

*** Failed ***
```
It appears that, even though I'm building for `TMS-1100` chapters aren't supported?  It should support a minimum of two chapters; four would be better for support of the `TMS-1400`.

mikeakohn, 2020-06-09:
I appreciate you reporting this to me.. but can you just give me a straight list (short summary) of what the problems are?

At this point I looked at the ldx one, and I do see an issue (assembling ldx 1 disassembles as ldx 0).  But to me this looks like what the correct line of code:

Line 133: c = tms1000_reverse_constant[opcode & 0x3] >> 2;

The docs I have say the immediate goes in 2 bits, reversed order.

For the other ones, I'm a little overwhelmed with all the text would like to go through them one by one, so a list would be really helpful.

ratsputin, 2020-06-09:
Apologies for the wall of text.  Most of what was going on I figured out over the course of the day and I updated the notes as I went through.

The basic issues with the disassembler are:

- `READ_RAM(a)` references the array `tms1000_address_to_lsfr` which only has 64 entries.  The value `a` can be anywhere from 0x0-0xfff.  The fix on Line 19 addresses this by looking up the offset within the page in the lsfr array then adding that to the page address.
- The `ldx` issue I referenced is from the TMS-1000 Programming Reference Manual page 8-13, section 8-7.1.  It's definitely a 3-bit value as the TMS-1000 series have a 3-bit RAM file index the X register is used to address.  Here's a [link](https://en.wikichip.org/w/index.php?title=File%3ATMS1000_Series_Programmer%27s_reference_manual.pdf&page=147) to the page from the docs
- The chapter and page calculations just stem from an incorrect number of bits being shifted.  The page identifier is four bits wide (pages 0-15) to give 16 pages per chapter.  The TMS-1100 has two chapters for 2K of ROM; the TMS-1400 has four chapters for 4K of ROM.  Those are the fixes for lines 201, 202, 279 and 280.  Here's a [link](https://en.wikichip.org/w/index.php?title=File%3ATMS1000_Series_Programmer%27s_reference_manual.pdf&page=21) to the page layout information for the TMS-1000/TMS-1200.  Here's a [link](https://en.wikichip.org/w/index.php?title=File%3ATMS1000_Series_Programmer%27s_reference_manual.pdf&page=115) for the TMS-1100/1300.

For the assembler, it seems to have three minor issues:

- As mentioned in the disassembler, the `ldx` processing is not done correctly
- In the listing, the assembler is generating the correct opcode; however, it is disassembling it incorrectly.
- The assembler itself doesn't seem to support chapters; it only allows for 16 pages of 64 bytes total assembly output (1K).  The TMS-1100 has 2K of ROM; the TMS-1400 has 4K of ROM.

Thanks for the quick reply!  I'm trying to document the internal workings of Dark Tower.

mikeakohn, 2020-06-09:
I see, TMS1100 is 3 bits:

http://www.nyx.net/~lturner/public_html/TMS1000ins.html

Line 133: c = tms1000_reverse_constant[opcode & 0x7] >> 1;

Fixes it.. there's also an issue in the assembler.

ratsputin, 2020-06-09:
I found a better example of the assembler issue.  The issue manifests itself when assembling the TMS-1100 sample `samples/tms1100/testing.asm`.

Code:
```
.tms1100

  a6aac
  a7aac
loop:
  clo
  br loop

  sbit 0
  rbit 1
  tbit1 1
  ldx 2
  call blah

  tcy 3
  tcmiy 3
  ldp 3
  ynec 3

blah:
  ynec 3
```
Listing output:
```
.tms1100

  a6aac

000 0/0/00: 7a a6aac                                    cycles: 6

  a7aac

001 0/0/01: 76 a7aac                                    cycles: 6

loop:
  clo

002 0/0/03: 0b br 0x03 (linear_address=0x02)            cycles: 6

  br loop

003 0/0/07: 83 ldx 4                                    cycles: 6

  sbit 0

004 0/0/0f: 30 mnea                                     cycles: 6

  rbit 1

005 0/0/1f: 34 mnea                                     cycles: 6

  tbit1 1

006 0/0/3f: 38 mnea                                     cycles: 6

  ldx 2

007 0/0/3e: 29 mnea                                     cycles: 6

  call blah

008 0/0/3d: fc mnea                                     cycles: 6

  tcy 3

009 0/0/3b: 4c mnea                                     cycles: 6

  tcmiy 3

00a 0/0/37: 6c mnea                                     cycles: 6

  ldp 3

00b 0/0/2f: 1c mnea                                     cycles: 6

  ynec 3

00c 0/0/1e: 5c mnea                                     cycles: 6

blah:
  ynec 3

00d 0/0/3c: 5c mnea                                     cycles: 6
```
Notice after the first couple of instructions are assembled and disassembled correctly, the assembler seems to lose its mind in the listing portion?  The assembler output (hex values) is correct but the disassembler is generating the wrong opcodes from those values.

Incidentally, in that specific sample, `clo` isn't a valid TMS-1100 opcode.  That's a TMS-1000 opcode.

mikeakohn, 2020-06-09:
The website above (what I based the assembler on) had clo as both TMS1000 and TMS1100.  But I see in the PDF I have clo isn't there. I'll take it out.

mikeakohn, 2020-06-11:
I hope that's the last of it... I kind of regret building this assembler, but it looked so easy from the one page I looked at and I thought I had something to test it on.

Thanks for helping out!

mikeakohn, 2020-06-16:
I'm assuming that took care of all the issues?  I'm going to go ahead and close this bug.

closed 2020-06-16 by mikeakohn