mikeakohn/naken_asm #84: call address misalignment

netlinker, 2020-06-30:
Dear Mike,

I'm currently working on a MSP430 project and I think I found an exotic bug.
To make it easier to find a code segment in disassembly and assembly, I tried to add an otherwise unused byte sequence before a program routine, like so:

byte_sequence:
  db 0x1,0x2,0x3

following_routine:
  add ....
  ....
  ....
ret

If I now try to call the routine by:

call #following_routine

the assembler does not use the start address of the "following_routine", but instead the address of the next free byte after "byte_sequence" which, in my case, happens to be an odd address.
Execution of this code totally stalls a MSP430F1101.

If I change it to:

byte_sequence:
  db 0x1,0x2,0x3,0x4

It is working perfectly well.

My suggestion would be to check all call addresses and align them to the next free even address.

mikeakohn, 2020-07-01:
I wrote a small program:

.msp430
  
main:
  db 0x13
  call #asdf

asdf:
  ret

and assembled it:

Pass 1...
Pass 2...
Warning: Instruction doesn't start on 16 bit boundary at msp430.asm:5.  Padding with a 0.

I'm curious what you did that you didn't get that warning.

netlinker, 2020-07-01:
Ahh, I see, I checked again and indeed I'm getting the same warning. I must have somehow overlooked it.

Is there any reason though why the call does not point to the padded address? In my opinion, calling an uneven address would not make sense anyway, at least on an msp430, or is there a use case I'm not aware of?

I may have a look at the source code myself soon to see how the assembler works under the hood, but the next week I'm relatively busy.

When my project is ready, I would gladly donate it to your list of msp430 examples after some cleaning, if there is any interest. It's a interrupt driven, full duplex software uart with a simple scpi parser.

hosewiejacke, 2020-07-02:
>Is there any reason though why the call does not point to the padded address?

One of my reasons for choosing assembly is the control of every byte.
I wouldn't like the assembler to do things I didn't tell him. YMMV.

netlinker, 2020-07-02:
I agree, that the assembler should automate as less, as possible, but if it fills in a byte for padding to align the program code (which I'm OK with, it even prints a warning), I would expect it to also modify the jump addresses to still point to the right position after padding.
The Problem is the following:

;; Main is still aligned to even addresses
main:
  mov #string,r6
  .....
  ;; Call the print routine to print stuff pointed to by r6
  call #print
ret

;; Insert an uneven number of bytes , quite common for strings
string:
  db  'H','e','l','l','o','!',0
;; Print would now start at an uneven address, but the assembler inserts a padding 0 to make it start on an even address.
print:
;; Print string here
  mov @r6, ......
  .....
  ;; And so on
ret

Now lets say print would start on an uneven address because of the uneven length of the string put in front of it.
Lets assume this is 0xf891. Now the "call #print" does point to the address 0xf891, which is what happens. Now the assembler checks for alignment and inserts a byte before the print routine to make it start on an even address, which would be 0xf892 then. The call instead is still pointing to 0xf891. In this case it does not matter if one likes the call address to be modified or not, or if the processor could actually work on uneven aligned code. The code will simply crash for the reason that command and argument are swapped. In my opinion, this should be changed.
As already said I find it a good feature that the assembler actually pads the data to align the code, but if it does so, it should make sure the call addresses get aligned, too. otherwise it could just state a big warning and leave the padding be all together, as the code won't run anyway.

mikeakohn, 2020-07-03:
I think MSP430 is the only assembler in naken_asm that even prints that warning.. and I put it there because someone previously asked for it. I agree with hosewiejacke though.. I think if the assembler starts fixing errors, it's asking for possible other issues.

I definitely appreciate the suggestion though.. if you have more let me know.

netlinker, 2020-07-03:
Ok then. I think in this case, case closed.
I'll definitely let you know when I have further suggestions.
Great Project and great work! I'd like to see it getting some traction, as it's the most easy to use assembler for the MSP430 in my opinion.

closed 2020-07-03 by netlinker