mikeakohn/naken_asm #87: Literal syntax

russells, 2021-02-08:
Hello,

I'm using recent naken_asm compiled from source.
```
$ git show
commit 343cdada86a847715572695c7303216970ad79e3 (HEAD -> master, origin/master, origin/HEAD)
Author: Michael Kohn <mike@mikekohn.net>
Date:   Fri Feb 5 20:37:12 2021 -0600
```

I tried to assemble some msp430 code with literal values, and got some unexpected results.

Here's a demo:

```
.msp430
.include "msp430/msp430x14x.inc"
.org 0x1100
        UTXRXE0_plus equ UTXE0 + URXE0
        UTXRXE0_or   equ UTXE0 | URXE0
        UTXRXE0_0xc0 equ 0xc0

;; All these should generate the same opcodes and immediate data.
        ; instruction                   ; generated code
        bis.b #UTXE0+URXE0, &ME1        ; bis.b #0x80, &0x0004
        bis.b #UTXE0|URXE0, &ME1        ; bis.b #0x40, &0x0004
        bis.b #UTXRXE0_plus, &ME1       ; bis.b #0x80, &0x0004
        bis.b #UTXRXE0_or  , &ME1       ; bis.b #0x40, &0x0004
        bis.b #(UTXE0+URXE0), &ME1      ; bis.b #0x80, &0x0004
        bis.b #(UTXE0|URXE0), &ME1      ; bis.b #0x40, &0x0004
        bis.b #(UTXRXE0_plus), &ME1     ; bis.b #0x80, &0x0004
        bis.b #(UTXRXE0_or  ), &ME1     ; bis.b #0x40, &0x0004
        bis.b #UTXRXE0_0xc0, &ME1       ; bis.b #0xc0, &0x0004 ; This one seems correct.
```

Assembling this gives no errors.  Disassembling it produces this:

```
naken_util - by Michael Kohn
                Joe Davisson
    Web: http://www.mikekohn.net/
  Email: mike@mikekohn.net

Version: February 5, 2021

Loaded 54 .text bytes from 0x1100
symbol              0x0000
symbol    testlit.s 0x0000
symbol              0x0000
Loaded elf testlit.elf from 0x1100 to 0x1135
Type help for a list of commands.

Addr    Opcode Instruction                              Cycles
------- ------ ----------------------------------       ------
0x1100: 0xd0f2 bis.b #0x80, &0x0004                     5
0x1102: 0x0080
0x1104: 0x0004
0x1106: 0xd0f2 bis.b #0x40, &0x0004                     5
0x1108: 0x0040
0x110a: 0x0004
0x110c: 0xd0f2 bis.b #0x80, &0x0004                     5
0x110e: 0x0080
0x1110: 0x0004
0x1112: 0xd0f2 bis.b #0x40, &0x0004                     5
0x1114: 0x0040
0x1116: 0x0004
0x1118: 0xd0f2 bis.b #0x80, &0x0004                     5
0x111a: 0x0080
0x111c: 0x0004
0x111e: 0xd0f2 bis.b #0x40, &0x0004                     5
0x1120: 0x0040
0x1122: 0x0004
0x1124: 0xd0f2 bis.b #0x80, &0x0004                     5
0x1126: 0x0080
0x1128: 0x0004
0x112a: 0xd0f2 bis.b #0x40, &0x0004                     5
0x112c: 0x0040
0x112e: 0x0004
0x1130: 0xd0f2 bis.b #0xc0, &0x0004                     5
0x1132: 0x00c0
0x1134: 0x0004
```

I see that in `include/msp430/msp430x14x.inc` there are lines like
```
TBCLGRP_3  equ 0x6000  ; TBCL0+TBCL1+TBCL2+TBCL3+TBCL4+TBCL5+TBCL6
```
implying this sort of thing isn't supported.  But then in `samples/msp430/guitar_proc.asm` there's
```
  mov.b #(USIDIV_0|USISSEL_2), &USICKCTL ; div 1, SMCLK
```

Is this literal syntax (`A+B` or `A|B`) supported?  Should every line in my example program be assembled to the same opcodes and immediate data?

Russell

mikeakohn, 2021-02-11:
Sorry for the slow reply... GitHub didn't send me an email about this and I just happened to log in (for other reasons) and saw this here.

So + and | are two different operations and they are supported of course.. the problem is in the include file:

URXE0      equ 0x40    ; USART0: UART receive enable
UTXE0      equ 0x40    ; USART0: UART transmit enable

Looks like the assembler is doing the right thing. Not sure where this include file came from... (yeah I can see I committed it)... not 100% sure what the correct values are.. looks like:

UTXE0 should be 0x80
URXE0 should be 0x40
USPIE0 should also be 0x40 for some reason.

I'm tempted to just delete this include file.

russells, 2021-02-12:
Apologies, I didn't think to check the include file.  The error in there should have been obvious from the results of + and |.

I'll feed  back any more that I find.

Nice work on this assembler, btw.  I had been struggling with gnu as and ld for a while, and thought that there must be a cleaner way to make hex files. And there is!  Thank you.

mikeakohn, 2021-02-12:
Thanks! That's exactly why I wrote it.. I couldn't find an assembler for MSP430 that I liked.

Definitely let me know if you find anything else or need something. I changed the include file for those 3 defines, but if those mistakes are in there I'm worried what else there is. I'm still tempted to delete it....

closed 2021-02-12 by mikeakohn