mikeakohn/naken_asm #88: msp430 "0(R3), R5" generates an extra 0 word in the output

jali2, 2021-02-24:
Assembling for msp430 instruction set:
    .msp430
    mov #1, R5        ; outputs 0x4315 as expected.
    mov 0(R3), R5    ; output 0x4315 0x0000 and that is incorrect.

Nothing with R3 as the source should generate a word after the instruction.

mikeakohn, 2021-02-25:
Hmm.. I think the assembler is doing exactly what you told it to do. Is there a valid reason to use the CG register like that?

If it really bothers you I could have the assembler print out a warning if anyone uses the r3 register?

jali2, 2021-02-25:
Hi, that was fast!

Sure, one of those "don't do that" things.Was researching using R3 as a destination reg, same thing, assembler errors on mov R5, #1, but not on mov R5, 0(R3).There are some Google results agreeing with you, don't do that, its a grey area, not well documented by TI.
I found it by accident when I ran a disassembler from zsquareplusc/python-msp430-tools over the hex output and got an "illegal instruction. Bit annoying that the *lst output does not show the extra 0x0000, but it is in the hex output.If perhaps the *lst should show the actual output generated, that would help.

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
zsquareplusc/python-msp430-tools

The project provides a number of scripts to work with the MSP430 embedded processor. Such as up/download firmwar...
 |

 |

 |

mikeakohn, 2021-02-25:
I modified the list file output for MSP430 so it should now disassemble all the data generated by an instruction.

closed 2021-02-25 by mikeakohn

jali2, 2021-02-25:
Nice, I will update and check it out later today.
That is some fast response time, impressive!

 
 

Closed #88.

mikeakohn, 2021-02-25:
I take support seriously.. :).