mikeakohn/naken_asm #91: 8051 ajmp/acall fail to assembly

wenij, 2021-04-14:
hi Sir,

the ajmp/acall in 8051 have issue, and it can't success assembly the simple code.
I check the 8051 source code, and it seems wrong in the OP_PAGE(page address) check.

I modify the code, and it seems can fix this issue.
Please check it.

BR,
Wenij
[8051_ajmp_fix.patch.zip](https://github.com/mikeakohn/naken_asm/files/6311149/8051_ajmp_fix.patch.zip)

mikeakohn, 2021-04-15:
Can you tell me what your "simple code" looks like so I can see what the issue is?

wenij, 2021-04-15:
you can try to compile this camel51.asm porting.
it should encounter this issue.
[camel51.asm.zip](https://github.com/mikeakohn/naken_asm/files/6318445/camel51.asm.zip)

mikeakohn, 2021-04-15:
Everything I've tried with ajmp on naken_asm and c51asm produces the same result or same error. Can you just post a small piece of code that you claim doesn't work so I can test it?

mikeakohn, 2021-04-15:
Okay, I got it:

.org 0xa000
main:
  ajmp main

mikeakohn, 2021-04-15:
I pushed a fix.. let me know if that takes care of for you.

wenij, 2021-04-15:
yes, correct. it solved the issue.
and thanks this great project.

mikeakohn, 2021-04-15:
Cool.. and thanks for letting me know.. helps me a lot :). Let me know if you find anything else.

closed 2021-04-15 by mikeakohn