mikeakohn/naken_asm #92: 8051 bit address assembly may need improve it

wenij, 2021-04-15:
hi Michael,

here is another 8051 bit  address assembly issue.
the valid bit address should be 0x20~0x2f , 0x80 ~ 0xff (%8 == 0)
and it can't accept number as bit address. only accept [number].bit

below is test code, 
==
.8051

setb 0x20.1		;should assembly as "d2 01", but as "d2 21"
setb 0x22.1		;should assembly as "d2 11", but as "d2 23"
setb 0x81.1 	; should report error, but assembly as "d2 82". since 0x81 are invalid bit address map
setb 0x10.1 	; should report error, but assembly as "d2 11", since 0x10 are invalid bit address map 
setb 0x20		;should accept it, and assembly as  "d2 20" ,but report error
==

BR,
Wenij

mikeakohn, 2021-04-16:
I guess I never needed 0x20.0 to 0x2f.7. Should be fixed now. I added new tests to tests/comparison/8051.txt which compares against the c51asm. Let me know if that takes care of it for you.

wenij, 2021-04-17:
just test it, and it still have issue.

below code will report error,
Error: Bit Address out of range (32,63) at t_bit_address.asm:15

;--- test code ---
.8051
.include "8051.inc"
#define  ri  SCON.0

.org 1000

setb 0x20.1		;should assembly as "d2 01", but as "d2 21"
setb 0x22.1		;should assembly as "d2 11", but as "d2 23"
;setb 0x81.1 	; should report error, but assembly as "d2 82". since 0x81 are invalid bit address map
;setb 0x10.1 	; should report error, but assembly as "d2 11", since 0x10 are invalid bit address map 
setb 0x20		;should accept it, and assembly as  "d2 20" ,but report error

.org 2000

cin:	jnb	ri, cin
	clr	ri
	mov	a, sbuf
	ret
;----

mikeakohn, 2021-04-17:
I have a file with:

    .8051
  
    .org 0x1000
    main:
      setb 0x20.1
      setb 0x20

./naken_asm -l 8051.asm

```
0x1000: d2 01      setb 0x20.1 [0x01]                       cycles:
  setb 0x20

0x1002: d2 20      setb 0x24.0 [0x20]                       cycles:
```

What am I doing differently that I don't see it?

wenij, 2021-04-17:
[image]
the issue should be in 
cin: jnb ri, cin

it can't be assembly

mikeakohn, 2021-04-17:
If you remove the .include "8051.inc" does it assemble?

wenij, 2021-04-17:
yes. it can assembly, now.
But why?

wenij, 2021-04-17:
so, the default 8051.inc have issue?

mikeakohn, 2021-04-17:
Everything defined in that file is defined in the assembler... I forgot why I did that. The other 8051 I used to test has those defines built in too.

wenij, 2021-04-17:
ok, now I remove the 8051.inc, it can  pass assembly.

another issue, it seems not warning or report error jf short range address out of range.
check below code,
;--
.8051

#define  ri  SCON.0

.org 0x1000
tt1:
	setb 0x20

.org 0x2000
	djnz	r1, tt1
	
;---

mikeakohn, 2021-04-17:
Do a pull and try now..

mikeakohn, 2021-04-17:
Is this issue fixed so I can close it now?

wenij, 2021-04-17:
Sorry,I am not able to test it now.
Will do it tomorrow.

wenij, 2021-04-18:
yes. it solved these issue. 
this issue can be close now.

mikeakohn, 2021-04-18:
Cool.. thanks!

closed 2021-04-18 by mikeakohn