mikeakohn/naken_asm #93: 8051 improve/optimize to try  acall/ajmp before direct assembly to lcall/ljmp

wenij, 2021-04-19:
some assembler ([AS31](https://github.com/ericponce/as31)) will try using acall/ajmp (even  coding it as lcall/ljmp), because it can save 1 bytes.
Can naken_asm(8051) have possible to optimize it like this way?

mikeakohn, 2021-05-14:
Sorry for a slow reply. This could be done, but this requires a possibly ugly change to the code since branching forward to a label that hasn't been defined yet would force it to fall back on the bigger instruction.

pulkomandy, 2022-08-14:
Hello,
I have a similar problem on unsp. The conditional jump instructions have only a 6-bit range, which is often not enough. The solution is to combine them with a goto instruction, so that:

    JE label

Becomes:

    JNE $+2
    GOTO label

It's a bit annoying to decide when this long form must be used (especially because I'm porting a C compiler and the compiler doesn't currently know about instruction sizes to make the decision)

Could this be handled by the optimizer?

mikeakohn, 2022-08-20:
I have the same issue working with Java Grinder also... although I have a "hint" distance in since it's translating from Java byte code.

It might be possible to do this, but it would take some work and would need to be done carefully since the assembler supports multiple platforms.