mikeakohn/naken_asm #97: Unknown instruction stm8

FrengerH, 2021-08-04:
This piece of stm8 binary
```
000000b0: c500 073c 0071 09c6 0152 08db 0333 068d  ...<.q...R...3..
000000c0: 0a34 0100 05ae 03ff 9490 ce80 aaae 80ac  .4..............
```
Disassembles to the following instructions.
```
0x80b0:  c5 00 07       bcp A, $7                                cycles=1
0x80b3:  3c 00          inc $00                                  cycles=1
0x80b5:  71             ???                                      cycles=?
0x80b6:  09 c6          rlc ($c6,SP)                             cycles=1
0x80b8:  01             rrwa X, A                                cycles=1
0x80b9:  52 08          sub SP, #$08                             cycles=1
0x80bb:  db 03 33       add A, ($333,X)                          cycles=1
0x80be:  06 8d          rrc ($8d,SP)                             cycles=1
0x80c0:  0a 34          dec ($34,SP)                             cycles=1
0x80c2:  01             rrwa X, A                                cycles=1
0x80c3:  00 05          neg ($05,SP)                             cycles=1
0x80c5:  ae 03 ff       ldw X, #$3ff                             cycles=2
0x80c8:  94             ldw SP, X                                cycles=1
0x80c9:  90 ce 80 aa    ldw Y, $80aa                             cycles=2
0x80cd:  ae 80 ac       ldw X, #$80ac                            cycles=2
```
As you see there is an unknown instruction. I expect the issue is that the 3c op-code is a two byte inc, instead of a one byte inc to which it's disassembled now. I did look into the code a bit, but I'm not sure how this can be detected since there is no prefix and both the 1 byte inc and two byte inc have the 3c op-code.

dlkeng, 2021-08-04:
Are you sure this is valid STM8 code and not data?

The PM0044 STM8 CPU Programming Manual (page 106) shows INC for "shortmem" (single byte) is the 3C opcode and the INC for "longmem" (two bytes) has prefix 72 and opcode 5C.
No opcode or prefix is shown in the manual for 71, therefore the unknown instruction.

mikeakohn, 2021-08-04:
I did a search through the manual also for 71 and there's only 3 instances.. none of them are for an instruction...

FrengerH, 2021-08-05:
You're right. I've been mistaken between the inc instruction with direct addressing and the inc instruction with pointers which also have the 3c opcode, but those both have a prefix. I'll close the issue since it's not an issue with the disassembler.

closed 2021-08-05 by FrengerH

mikeakohn, 2021-08-05:
Cool... if you see anything else, please bring it up so it can be looked into. I'd like the assembler / disassembler to be as bug-free as possible.