mikeakohn/naken_asm #100: Broken .ifdef/.else/.endif

elevendroids, 2021-11-01:
Hi,

After upgrading the assembler to the current revision (as of 1 November 2021), I've noticed that the `ifdef`/`else`/`endif` statements no longer work as expected - seems like `else` is kinda half-broken :wink:
I've narrowed down the issue to the refactoring from the commit df2b4c59b6875f6edae41188fe62b03f47cbdd46

Example:

```asm
.stm8

#define WANT_LESS_FOO 1

#ifdef WANT_LESS_FOO
#define FOO 1
#else
#define FOO 2
#endif

org 0x8000
    int FOO
```

Output:
```
[michal@silverbox naken_asm-bug]$ make
NAKEN bug.asm -> bug.hex

naken_asm

Authors: Michael Kohn
         Joe Davisson
    Web: http://www.mikekohn.net/
  Email: mike@mikekohn.net
Version: May 23, 2021

 Input file: bug.asm
Output file: bug.hex
  List file: bug.lst

Pass 1...
Error: Macro 'FOO' already defined.
Error: unmatched .endif at bug.asm:0
** Errors... bailing out

Program Info:
Include Paths: .
               lib/stm8-asm-lib/include
               /usr/local/share/naken_asm/include
               include
 Instructions: 0
   Code Bytes: 0
   Data Bytes: 0
  Low Address: fffffffe (4294967294)
 High Address: 0000 (0)

*** Failed ***

make: *** [Makefile:28: bug.hex] Error 1
```

When I comment out the `#define WANT_LESS_FOO 1` statement - the "code" assembles fine.

This fails in a bit different way with `equ` - looks like second (incorrectly parsed) identifier gets expanded to the previously defined value.
Example:

```asm
.stm8

.define WANT_LESS_FOO 1

.ifdef WANT_LESS_FOO
FOO equ 1234
.else
FOO equ 2345
.endif

org 0x8000
    int FOO
```

Output:
```
[michal@silverbox naken_asm-bug]$ make
NAKEN bug.asm -> bug.hex

naken_asm

Authors: Michael Kohn
         Joe Davisson
    Web: http://www.mikekohn.net/
  Email: mike@mikekohn.net
Version: May 23, 2021

 Input file: bug.asm
Output file: bug.hex
  List file: bug.lst

Pass 1...
Error: Unknown instruction '1234equ' at bug.asm:8
** Errors... bailing out

Program Info:
Include Paths: .
               lib/stm8-asm-lib/include
               /usr/local/share/naken_asm/include
               include
 Instructions: 0
   Code Bytes: 0
   Data Bytes: 0
  Low Address: fffffffe (4294967294)
 High Address: 0000 (0)

*** Failed ***

make: *** [Makefile:28: bug.hex] Error 1
```

Again, after commenting out the `.define` statement, only the `.else` part gets parsed and the thing assembles fine.

With the previous commit (cf91bef3c093fa548052640b17894fc8985cdd56) everything works as expected.

mikeakohn, 2021-11-03:
Sorry I haven't had a chance to look into this yet... hopefully tomorrow night.

mikeakohn, 2021-11-05:
I pushed a fix, but there's two things before I'm going to close the issue:

1) Need to look into the return values from some of those functions, maybe put some enums in there so it reads cleaner.
2) Needs to have a test so that doesn't happen again.

Thanks for letting me know! I'll probably get to that Sunday. It should be usable now though...

elevendroids, 2021-11-05:
I gave it a quick check - works a treat :+1: :slightly_smiling_face: 

Yeah, a regression test for this (and similar basic syntax elements) is probably a good idea - should prove useful, especially when refactoring/cleaning up stuff.

Thanks for fixing this!

mikeakohn, 2021-11-08:
Added the unit test (will do the other stuff in another commit)... thanks again for letting me know and let me know if you find anything else..

:)

closed 2021-11-08 by mikeakohn