mikeakohn/naken_asm #101: configure script for just one CPU fails

nils-eilers, 2021-12-01:
Tried to build a 8048 only assembler. Building an 6502 only assembler fails as well. Didn't try other CPUs.

Compilation output:

`[j@xps15 naken_asm]$ ./configure --enable-8048

Configuration:
  Install to: /usr/local
            : /usr/local/bin, /usr/local/share/naken_asm/include
    Compiler: gcc
     LDFLAGS:  -s
UTIL LDFLAGS:  -lreadline
      CFLAGS: -Wall -DREADLINE -O3 -DNO_MSP430
     DEFINES:  -DENABLE_8048
INCLUDE_PATH: /usr/local/share/naken_asm/include
        BINS: naken_asm, naken_util, naken_prog

Now type: make

[j@xps15 naken_asm]$ make      
make[1]: Entering directory '/home/j/Projekte/software/asm/naken_asm/build'
gcc -c ../asm/common.c -o asm/common.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../asm/8048.c -o asm/8048.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../disasm/8048.c -o disasm/8048.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/add_bin.c -o common/add_bin.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/assembler.c -o common/assembler.o -Wall -DREADLINE -O3 -DNO_MSP430 -DENABLE_8048 -I..
gcc -c ../common/cpu_list.c -o common/cpu_list.o -Wall -DREADLINE -O3 -DNO_MSP430 -DENABLE_8048 -I..
gcc -c ../common/directives.c -o common/directives.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
../common/assembler.c: In function 'assembler_init':
../common/assembler.c:40:3: warning: implicit declaration of function 'configure_cpu' [-Wimplicit-function-declaration]
   40 |   configure_cpu(asm_context, 0);
      |   ^~~~~~~~~~~~~
gcc -c ../common/directives_data.c -o common/directives_data.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/directives_if.c -o common/directives_if.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/directives_include.c -o common/directives_include.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/eval_expression.c -o common/eval_expression.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/eval_expression_ex.c -o common/eval_expression_ex.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/ifdef_expression.c -o common/ifdef_expression.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/imports_ar.c -o common/imports_ar.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/imports_get_int.c -o common/imports_get_int.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/imports_obj.c -o common/imports_obj.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/linker.c -o common/linker.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/print_error.c -o common/print_error.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/macros.c -o common/macros.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/memory.c -o common/memory.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/memory_pool.c -o common/memory_pool.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/symbols.c -o common/symbols.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/tokens.c -o common/tokens.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../common/var.c -o common/var.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
gcc -c ../simulate/null.c -o simulate/null.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
make[1]: *** No rule to make target 'simulate/8048.o', needed by 'default'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/j/Projekte/software/asm/naken_asm/build'
make: *** [Makefile:13: default] Error 2
[j@xps15 naken_asm]$ 
`

mikeakohn, 2021-12-01:
I pushed a commit that should fix it. I tested 8048 and it seems to assemble (even without the .8048 directive). Let me know how it goes...

oh.. and thanks for letting me know :)

nils-eilers, 2021-12-01:
Works for me! Many thanks! E-mail with further notes sent :)

closed 2021-12-01 by nils-eilers

mikeakohn, 2021-12-03:
I sent a response to your email a couple days ago.. just wanted to make sure you got it.

nils-eilers, 2021-12-03:
No, I didn't receive anything, spam folder checked as well. Could you please re-send it to nils.eilers@gmx.de ? And thanks for asking!

thesilicon1, 2024-04-29:
For today doesn't worked compiling for one or more

mikeakohn, 2024-04-29:
This feature is broken for you? I was actually thinking this weekend of getting rid of it altogether, it doesn't seem useful.