mikeakohn/naken_asm #108: naken_util disassembler for unsp is a bit confused

pulkomandy, 2022-08-13:
Here are a few oddities I noted:
- The print command takes addresses in bytes, but the disasm command takes addresses in words. I think they should both use words
- The print16 command prints things in swapped endian, so for example the address 55 08 gets printed as 0855 (not sure if other things like opcodes in disassembly output, etc are also affected)

pulkomandy, 2022-08-13:
- jne instruction is decoded as `cmp pc, #constant`

mikeakohn, 2022-08-13:
The jump instructions should be fixed in the disassembler.

The print and write commands are super awkward for this chip (and really any chip with multiple bytes per address). I think I have it working better than it was before though:

    0x0000: 04cb add r2, [r3--]
    0x0001: 4e42 jne 0x0000 (offset=-2)
    0x0002: 4e00 jne 0x0003 (offset=0)

    stopped> print16 1
    0x0001: 4e42 4e00 0000 0000 0000 0000 0000 0000 BN.N............
    0x0009: 0000 0000 0000 0000 0000 0000 0000 0000 ................

Let me know...

closed 2022-08-13 by mikeakohn