mikeakohn/naken_asm #109: Feature request: Support for Padauk

kimstik, 2022-10-11:
Subj,
Only this kernel is not supported from the supported by SDCC.

ref: https://github.com/free-pdk/fppa-pdk-tools

kimstik, 2022-10-11:
not so different :)
for example instruction "CALL K" is encoded as:
PDK13: 111kkkkkkkkkk 
PDK14: 111kkkkkkkkkkk
PDK15: 111kkkkkkkkkkkk

mikeakohn, 2022-10-11:
Kind of looks like this could piggy-back on top of PIC code (in naken_asm that is).. so it should be pretty easy to add all of these. I think I was actually thinking of doing this once but I couldn't find real chips to test this on...

I'll add this to my list.. maybe I'll start it in the next couple weeks.

kimstik, 2022-10-13:
It's true they are very similar to the PICs. 
I can share what I have for testing.
More complex task is the programmer. stm32 which is used there is not so easy to buy.

mikeakohn, 2022-10-13:
So the names of the chips in the assembler would be .pdk14 .. so writing code would be:

    .pdk14
    swap a

etc?

mikeakohn, 2022-10-13:
Do you have a small pdk14 you can send me? assembled already would be better, but source code would be helpful too... even if it's just 7 or 8 instructions.

mikeakohn, 2022-10-14:
I added the tables for those 4 chips... I'm pretty sure there are errors all over the 4 pages I used to build those tables:

    https://free-pdk.github.io/instruction-sets/PDK13
    https://free-pdk.github.io/instruction-sets/PDK14
    https://free-pdk.github.io/instruction-sets/PDK15
    https://free-pdk.github.io/instruction-sets/PDK16

There are instructions that don't fit a pattern... for example PDK16:

    comp A, M
    comp M, A

then starting at the "add" the pattern switches to:

    add M, A
    add A, M

Instruction sets usually don't do this. There are other oddities too with the opcode on the left not matching the bit encodings.

mikeakohn, 2022-10-14:
All 4 assemblers are now in naken_asm. My guess is they are buggy and need some tests.

kimstik, 2022-10-16:
I would like to test it a bit. May I ask you to release windows binaries?

kimstik, 2022-10-16:
BTW nice test candidate:
https://github.com/free-pdk/easy-pdk-showcase-projects/tree/master/noiseplug

mikeakohn, 2022-10-16:
I can, but it's a mess. There are several instructions that have a single name but different operands. Like "xor" has:

    xor a, M
    xor a, k

both M and k are a number.. so I guess for the assembler to know that k is supposed to be a constant it should be xor a, #k ? But goto k doesn't do the #k? and then there is

    t0sn M.b
    t0sn IO.b

both M and IO are a number.. to the assembler they look the same. So I've been testing against the assembler in SDCC... that one seems to make up its own syntax using t0sn.io IO, #bit

SDCC I think also is buggy.. if I do SDCC's assembler for pdk13:

    goto 1

It comes back as: 0x1802 which should disassemble as goto 2

SDCC's pdk16 assembler assembles "goto 1" as 0x3001 which looks correct.

Does Padauk not have a data sheet with all the correct opcodes and syntax and such? It's possible the only way to make sure this is correct is to use their official IDE?

mikeakohn, 2022-10-19:
I compiled and posted a new version of naken_asm.

mikeakohn, 2022-10-27:
I need your opinion... so the syntax for say:

    add M, a

the free pdk tools disassembles it as add [address], a  .. SDCC uses I think add address, a .. and anywhere there is a constant for those instructions it uses # constant.

then there is instruction like:

    xor.io 0x03, a
    xor [0x02], a
    xor a, [0x02]
    xor a, 0x2c

SDCC appends .io to instructions that are for IO.
Padauk requires the name of the IO to be used to it's obvious. 
naken_asm could follow SDCC or just use "xor 0x03, a"

What syntax do you like better?

kimstik, 2022-10-28:
xor.io is a correct way.
SDCC did not immediately come to this point. They originally used a raw "xor 0x03, a" notation.
ref: 2021/07/26 Visenri https://sourceforge.net/p/sdcc/bugs/3259/

mikeakohn, 2022-10-28:
I wonder why the PDK free tools didn't do that too? I'll go with xor.io .. it's kind of easier anyway. The more I was thinking about it, I think I like the [address] thing better than the # constant, so I'm going to go that way.

I was able to build test code from the Padauk tools for PDK13 and PDK15, so I think by Monday I could have it as a working Beta.

mikeakohn, 2022-10-29:
I added tests for pdk13 and pdk15 that test against Paduak's assembler. Those are all passing now so they should be good. I'll try to finish pdk14 and pdk16 this weekend.

mikeakohn, 2023-12-25:
This should be complete.

closed 2023-12-25 by mikeakohn