mikeakohn/naken_asm #118: EQU crash and weirdness

erkkah, 2023-01-08:
I found some interesting things while porting old code containing `EQU`.

## 1

This example makes the assembler crash with a stack overflow:

```assembler
    XA  EQU R1
    XA  EQU R1
    mov XA, #0
```

In the real world example, there is more code between the repeated `equ` - statements.
I guess the second statement expands to `R1 EQU R1` which causes the assembler to enter that spiral.

Not sure what the best solution is here. If it's expected behaviour to have macro expansions on the lvalue - side, then the recursion needs to be detected. 

## 2

Replacing `EQU` with `.define` does not cause a crash, the duplicate definition is detected and logged, but the assembler reports success. Maybe an error that is not passed up?

## 3

If there is only a single space between `XA` and `EQU` in the example, another weird thing happens:

```
Error: Unknown instruction 'R1equ' at ...
```

erkkah, 2023-01-08:
Oh, BTW, I'm happy to look into these with some guidance from you. I'm in the middle of other things and wanted to report this while I remember.

mikeakohn, 2023-01-08:
Hmmm.. that should really be an error since XA was already defined?

I'd actually prefer to make changes / fixes to the core of the assembler myself... is this blocking you?

mikeakohn, 2023-01-08:
.msp430
    xa equ r2
    xa equ r3
    mov #0, &xa

This seems to bomb out with an error (not a duplicate though). It would have to do a look-ahead to figure out the next token is equ which would probably not be very efficient.

erkkah, 2023-01-08:
Not blocking, but it would be neat to find some kind of workaround, being able to re-define or un-define would help.
There is a bit of register setup in front of functions, and several functions use the same defines.

If there is no other way to solve it now, I can just use plain old documentation.

erkkah, 2023-01-08:
One solution I came up with is to wrap each function in a macro which is directly expanded.
Then I can use the macro parameters as local defines. :)