mikeakohn/naken_asm #119: 8051 to support bit addressed by number

cjacker, 2023-01-21:
**Problem:**

naken_asm doesn't support bit address by number. 

Such as:
```
MOV C, 19H
```

`19H` here means the 19th bit, it's equal to `23H.1`

**Description:**

There are 2 kinds of bit addr notation in 8051 asm.

- dot notation, such as '20H.1'   '80H.1', the dot works as offset
- by bit number,
   * besides the dot notation, bit can be addressed by bit number.
   * bits within 0x20h - 0x2Fh is numbered from 00H to 7FH.
   * bits above 0x80h RAM (acctually bits of the bit addressable SFR)
      is numbered from 80H to FFH, and each bit number is exactly 
      equal to the start addr of the byte + bit offset of the byte. 
      thus bit 087H is as same as 080H.7

bit number is mentioned in Intel MCS-51 manual:

[image] 

Both of the 'dot notation' and 'number notation' should support SLASH notation, such as '/87H.0' or '/87H'.

**Proposal:**
Below instructions are all instructions related to bit operation,
and should support both of the 'dot notation' and 'number notation'.

| opcode | instruction                  |
|--------|------------------------------|
| 10h    | JBC bit addr, code addr      |
| 20h    | JB bit addr,code addr (rel)  |
| 30h    | JNB bit addr,code addr (rel) |
| 72h    | ORL C,bit addr               |
| 82h    | ANL C,bit addr               |
| 92h    | MOV bit addr,C               |
| A0h    | ORL C,/bit addr              |
| A2h    | MOV C,bit addr               |
| B0h    | ANL C,/bit addr              |
| B2h    | CPL bit addr                 |
| C2h    | CLR bit addr                 |
| D2h    | SETB bit addr                |

Here is a proposal patch to fix these problem:

```
diff -Nur naken_asm/asm/8051.c naken_asm.fix/asm/8051.c
--- naken_asm/asm/8051.c	2022-06-08 14:04:15.000000000 +0800
+++ naken_asm.fix/asm/8051.c	2023-01-21 23:15:46.464990030 +0800
@@ -34,6 +34,7 @@
   OPERAND_AT_DPTR,
   OPERAND_DATA,
   OPERAND_NUM,
+  OPERAND_SLASH_NUM,
   OPERAND_SLASH_BIT_ADDRESS,
   OPERAND_BIT_ADDRESS,
 };
@@ -83,6 +84,18 @@
   return -1;
 }
 
+static int is_bit_op(int op)
+{
+  if(op == 0x10 || op == 0x20 ||
+     op == 0x30 || op == 0x72 ||
+     op == 0x82 || op == 0x92 ||
+     op == 0xa0 || op == 0xb0 ||
+     op == 0xa2 || op == 0xb2 ||
+     op == 0xc2 || op == 0xd2)
+    return 1;
+  return 0;
+}
+
 int parse_instruction_8051(struct _asm_context *asm_context, char *instr)
 {
   char instr_case_mem[TOKENLEN];
@@ -240,7 +252,9 @@
           else
         {
           tokens_push(asm_context, token, token_type);
-          operands[operand_count].type = OPERAND_NUM;
+          // using OPERAND_SLASH_NUM here to get the correct op
+          // avoid confusing with OPERAND_NUM and lost slash.
+          operands[operand_count].type = OPERAND_SLASH_NUM;
         }
       }
 
@@ -385,17 +399,14 @@
             break;
           case OP_SLASH_BIT_ADDR:
             if ((operands[r].type != OPERAND_SLASH_BIT_ADDRESS &&
-                 operands[r].type != OPERAND_NUM) ||
+                 operands[r].type != OPERAND_SLASH_NUM) ||
                 (operands[r].value < 0 ||
                  operands[r].value > 255)) { r = 4; }
             break;
           case OP_BIT_ADDR:
             if (operands[r].type != OPERAND_BIT_ADDRESS)
             {
-              // This seems kind of invalid, but some other assembler allows
-              // the bit address to be defined as the binary version of the
-              // address rather than address.bit for at least clr and setb.
-              if (n == 0xd2 || n == 0xc2)
+              if (is_bit_op(n))
               {
                 if (operands[r].type != OPERAND_NUM) { r = 4; }
               }
@@ -486,7 +497,7 @@
             }
             case OP_SLASH_BIT_ADDR:
             {
-              if (operands[r].type == OPERAND_NUM)
+              if (is_bit_op(n) && operands[r].type == OPERAND_SLASH_NUM)
               {
                 num = operands[r].value;
               }
@@ -506,7 +517,7 @@
             }
             case OP_BIT_ADDR:
             {
-              if ((n == 0xd2 || n == 0xc2) && operands[r].type == OPERAND_NUM)
+              if (is_bit_op(n) && operands[r].type == OPERAND_NUM)
               {
                 num = operands[r].value;
               }
```

- The reason to introduce a 'OPERAN_SLASH_NUM' :
   Since naken_asm only special treat the CLR/SETB instructions before, there is no conflict    between `72H: ORL C, bit addr` and `A0H: ORL C,/bit addr`.  After supporting bit addr by number, they are indistinguishable if  both use OPERAND_NUM, and will get a wrong opcode for `ORL C, /bit`.

**Test code:**
Below sample code should compiled without any problem and get the exactly same result as Keil A51.

```
; All instructions related to bit addressed by number
; comment out '.8051' to build with A51.

.8051
ORG 0000H
MAIN:
    ; below 2 line is exactly same.
    MOV C, 19H
    MOV C, 23H.1

    JBC 80H.7, TEST
    JBC 87H, TEST

    JB  80H.7, TEST
    JB  87H, TEST

    JNB 80H.7, TEST
    JNB 87H, TEST

    ORL C, 80H.7
    ORL C, 87H

    ANL C, 80H.7
    ANL C, 87H

    ORL C, /80H.7
    ORL C, /87H

    ANL C, /80H.7
    ANL C, /87H
	
    MOV 80H.7, C
    MOV 87H, C

    MOV C, 80H.7
    MOV C, 87H

    CPL 80H.7
    CPL 87H
	
    CLR 80H.7
    CLR 87H
	
    SETB 80H.7
    SETB 87H

    SJMP $

    TEST:
	NOP
END
```

cjacker, 2023-01-22:
**Related issue:**

There is still a little problem left not fixed, for example, we can NOT distinguish `P0` (80h)  from P0.0(the `80h` bit) and `80H` in bit addr related instructions.

For example, `setb P0` will be compiled without any problem, it should NOT.

`P0` was usually defined using `equ 80h` in include file, and it is really a valid number and substituted with `80h` when symbol lookup in `tokens_get`, `tokens_get` will always return a number for such SFRs. there is no change to determine it is the string symbol 'P0' or it's `P0.0` and `80H`.

I tried to find a way to fix it, and it should NOT affect the codes of other architectures. the goal is to find out token '128' is P0 or P0.0 or 80H.

Here is a proposal to fix it:
- Add two more fields in `_asm_context` struct to indicate it's a 8051 bit addressable SFR and the original SFR symbol string (4 byte string should be enough), the symbol string only be used when print an error. these two fields will never use by other architectures, thus it will NOT affect the codes of other archs.
- If the following token is a dot, clear the sfr flag.
- Before `add_bin8`, if it is bit_op and the sfr flag is set, throw an error and quit.

Here is the patch, it's just a try, may be not good enough since it has to touch `tokens.c`:

```
diff -Nur naken_asm/asm/8051.c naken_asm.sfr-bit-addressable/asm/8051.c
--- naken_asm/asm/8051.c	2023-01-22 19:31:43.387640283 +0800
+++ naken_asm.sfr-bit-addressable/asm/8051.c	2023-01-22 19:32:23.579703988 +0800
@@ -256,6 +256,7 @@
         ignore_operand(asm_context);
         operands[operand_count].value = 0x80;
         operands[operand_count].type = OPERAND_SLASH_BIT_ADDRESS;
+        asm_context->is_8051_bit_addressable_sfr = 0;
       }
         else
       {
@@ -277,6 +278,7 @@
 
           operands[operand_count].type = OPERAND_SLASH_BIT_ADDRESS;
           operands[operand_count].bit = num;
+          asm_context->is_8051_bit_addressable_sfr = 0;
         }
           else
         {
@@ -314,6 +316,7 @@
             operands[operand_count].value = 0x80;
             operands[operand_count].bit = 0;
             operands[operand_count].type = OPERAND_BIT_ADDRESS;
+            asm_context->is_8051_bit_addressable_sfr = 0;
           }
         }
 
@@ -339,6 +342,7 @@
 
           operands[operand_count].type = OPERAND_BIT_ADDRESS;
           operands[operand_count].bit = num;
+          asm_context->is_8051_bit_addressable_sfr = 0;
         }
           else
         {
@@ -541,6 +545,11 @@
 
               if (num == -1) { return -1; }
 
+              if (is_bit_op(n) && asm_context->is_8051_bit_addressable_sfr) {
+                print_error_unexp(asm_context->orig_8051_sfr_string, asm_context);
+                return -1;
+              }
+
               add_bin8(asm_context, num, IS_OPCODE);
               count++;
               break;
@@ -561,6 +570,11 @@
 
               if (num == -1) { return -1; }
 
+              if (is_bit_op(n) && asm_context->is_8051_bit_addressable_sfr) {
+                print_error_unexp(asm_context->orig_8051_sfr_string, asm_context);
+                return -1;
+              }
+
               add_bin8(asm_context, num, IS_OPCODE);
               count++;
               break;
diff -Nur naken_asm/common/assembler.h naken_asm.sfr-bit-addressable/common/assembler.h
--- naken_asm/common/assembler.h	2023-01-09 13:15:39.000000000 +0800
+++ naken_asm.sfr-bit-addressable/common/assembler.h	2023-01-22 19:07:57.729380725 +0800
@@ -81,6 +81,8 @@
   uint8_t in_repeat : 1;
   uint32_t flags;
   uint32_t extra_context;
+  uint8_t is_8051_bit_addressable_sfr;
+  char orig_8051_sfr_string[8];
 };
 
 void assembler_init(struct _asm_context *asm_context);
diff -Nur naken_asm/common/tokens.c naken_asm.sfr-bit-addressable/common/tokens.c
--- naken_asm/common/tokens.c	2023-01-09 13:15:39.000000000 +0800
+++ naken_asm.sfr-bit-addressable/common/tokens.c	2023-01-22 19:14:52.627038321 +0800
@@ -650,6 +650,40 @@
 
   if (token_type == TOKEN_STRING)
   {
+    // for 8051.
+    //
+    // tokens_get will finally return a number 128(0x80h) 
+    // after symbol lookup when encounter a symbol such as 'P0'.
+    //
+    // to support bit address by bit number, we need to
+    // distinguish the 128 of 'P0' from 'P0.0' and '0x80h'.
+    //
+    // Here when encounter all known bit addressable SFR,
+    // record the original symbol string and 
+    // mark is_8051_bit_addressable_sfr flag as 1.
+    //
+    // the flag will be cleared in asm/8051.c if it followed by a dot,
+    // then we can distinguish 'P0.0' from 'P0' for bit number.
+    //
+    // if finally the flag is not cleared before add_bin8, 
+    // that means it is a SFR instead of dot notation, then throw an error.
+
+    if (strcasecmp(token, "P0") == 0 ||
+        strcasecmp(token, "TCON") == 0 ||
+        strcasecmp(token, "P1") == 0 ||
+        strcasecmp(token, "SCON") == 0 ||
+        strcasecmp(token, "P2") == 0 ||
+        strcasecmp(token, "IE") == 0 ||
+        strcasecmp(token, "P3") == 0 ||
+        strcasecmp(token, "IP") == 0 ||
+        strcasecmp(token, "PSW") == 0 ||
+        strcasecmp(token, "ACC") == 0 ||
+        strcasecmp(token, "B") == 0)
+    {
+      asm_context->is_8051_bit_addressable_sfr = 1;
+      strcpy(asm_context->orig_8051_sfr_string, token);
+    }
+
     int param_count = 0;
     char *macro = macros_lookup(&asm_context->macros, token, &param_count);
     uint32_t address;
```

**Test cases:**
```
P0 EQU 80H
```

Each of the below instructions will print an error: `Error: Unexpected token 'P0' at ...`
```
  ANL C, P0
  ANL C, /P0
  CLR P0
  CPL P0
  JB  P0, TEST
  JBC P0, TEST
  JNB P0, TEST
  MOV P0, C
  MOV C, P0
  ORL C, P0
  ORL C, /P0
  SETB P0
```

mikeakohn, 2023-01-23:
I haven't had time to look at this.. I'll look at it tomorrow night hopefully. I believe the assembler used to support what you want when I first added 8051 and I ended up taking out.

cjacker, 2023-01-23:
Thanks for reply.

It's not a big issue, take a look when you are available.

I didn't try history codes. Maybe before the hardcode SFRs removed, addressing bit by number could be supported. Now, there is only a hack left for setb/clr.

By the way, should it have special directives such as 'BIT' to distinguish SFR ,BYTE and BIT definitions? it will make things easier.

mikeakohn, 2023-01-29:
Sorry for the slow response, busy week.

Thanks for letting me know about this. It's probably a little easier for me if you just show the instructions that don't work... don't really need the documentation (I have it) or code. The assembler did have support for bit addressing, it just had some bugs.

I added new tests to:

    tests/comparison/template/8051.txt

.. and verified there was a problem. This test compares against the c51asm assembler and looks like it matches. If it looks good for you, please close the issue or if not let me know which instructions aren't assembling properly.

And thanks again for the help by reporting this! :)

cjacker, 2023-01-30:
Thanks.

It fixed the 'bit addressing by number' issue, but did not fix the 'related issue' that standard SFR name can be used with bit related instructions.

For example, below codes should not get compiled (c51asm assembler failed to compile):
```
.8051
ORG 0000H
P0 EQU 80H  ; or a standard include file.
MAIN:
  ANL C, P0
END
```

It is not a syntax error since `P0` (80h) is a valid value for bit addr, the assembler can not distinguish it from `P0.0`. but it is  a semantic error and should not be supported.

Maybe it's hard to fix, I posted a dirty fix in comment. as I mentioned before if naken_asm have special directives such as 'BIT', 'SFR' to define BIT or SFR, it will make things easier.

mikeakohn, 2023-02-06:
Do people actually use P0 there instead of P0.0? I tried this with the c51asm and it gives an error.

mikeakohn, 2023-02-12:
So is this really a problem or can this be closed?

mikeakohn, 2023-02-17:
I believe this is working as expected, so I'm going to go ahead and close this.

closed 2023-02-17 by mikeakohn