mikeakohn/naken_asm #120: TMS34010 - SP register not handled correctly

deckarep, 2023-02-10:
Hello!

Cool project, I wanted to report that for the TMS34010 CPU it looks like the `stack pointer` isn't handled exactly correct on at least the `MOVI` instruction with [IL] (immediate longs) passed. Potentially other instructions will also be affected when referencing the stack pointer (`SP`) register.

Example

```
.tms340
.little_endian
.org 0xFF810F80 

.set foo = 0x010ffff0 ;top of stack

START:
    movi    foo, sp, l
```

Expected:
```
EF09 F0FF 0F01
```

Got:
```
FFFF F0FF 0F01
```

deckarep, 2023-02-10:
Upon further inspection I think I've zeroed in on where the bug is manifesting. There is a related comment in the code about the `SP` register being associated with either register file which is defined as either Register File `A = 0`, or Register File `B = 1`.

When this related code runs: https://github.com/mikeakohn/naken_asm/blob/master/asm/tms340.c#L452

If you have a register operand, your register file comes back as `-1` .

Then later when the case statement runs for handling either an `OP_RD`, `OP_RDS` and likely other register related operand types a register value of `-1` cause the opcode to not be packed correctly in the final opcode result.

Basically this code comes out wrong:

```
opcode |= operands[i].reg;
opcode |= operands[i].r << 4;
opcode |= operands[i].reg << 5;
```

I'm not yet clear on where the fix should be placed, but I did confirm that if the code can return a proper register file `0 or 1` for the stack pointer `SP` register the binary encoding will be correct.

mikeakohn, 2023-02-10:
Thanks for the heads up.. I think I got it fixed.

Oddly, a couple weeks ago I was talking with one of my friends about regretting spending time adding TMS34010 to naken_asm... felt like a total waste of time since I didn't think anyone would use it. So despite being a bug, it was kind of cool getting notified about this :). 

If this fixes it for you, please close the bug.. and if you find anything else let me know. Also, I'd still be really curious to see if you do something with the assembler.

Thanks again!

deckarep, 2023-02-11:
I found one additional thing so far unrelated to this issue but related to the use of one of the align directives. Perhaps I used it wrong but it caused the assembler to go into an infinite loop generating tons of zeroes until I killed the program. My file with only dozens of lines of assembly grew to be 250MB with mostly zeroes. I’ll see if I can repro….and post a new issue.

deckarep, 2023-02-11:
Also, I’ll keep banging on the tms340 code to see if I can help identify anything further. I appreciate your quick response and fix.

deckarep, 2023-02-11:
One more question: do you know where I can get the original assembler to do some additional testing and comparisons with what naken does?

mikeakohn, 2023-02-11:
I used gspa (saved the assembler name in a script.. good on me).. of course I can't figure out where I put it. I was going to add movi 0x010ffff0,sp, l to it so it won't break again, but I lost it. Checked a bunch of docker containers and VM's and don't see it.

deckarep, 2023-02-13:
@mikeakohn  - i confirmed this looks good. Thanks for the fix...also I tracked down a version of the (hopefully official) TMS34010 assembler called: GSPA.EXE which looks to work correctly. *Although*, there is a newer version that I'm keeping my eyes peeled for.

You can get it here: https://github.com/historicalsource/revolution-x/blob/main/GSPA.EXE

I'm sure you're aware that many arcade games by Midway including Mortal Kombat and NBA JAM used this processor/assembler.

closed 2023-02-13 by deckarep

mikeakohn, 2025-01-25:
> Is there a sample of tms340 for usage with Mortal Kombat? [@mikeakohn](https://github.com/mikeakohn) have you used this for patching MK?

I actually haven't had a chance to do TMS340 yet... if you do anything with it it would be cool to see...

mikeakohn, 2025-01-26:
Macros are supported... but a lot of that other syntax isn't. But a macro could probably fix some of the syntax differences.. or a Python script that converts that syntax to naken syntax.

mikeakohn, 2025-01-30:
Those directives don't exist in naken_asm and if they are needed would probably need to be added.

mikeakohn, 2025-01-30:
If you want a new feature, you can file an issue and I can see about adding it. Also, I think those fields are only valid on writing out to ELF.. maybe MachO formats.