mikeakohn/naken_asm #122: 65C816 Direct Page Indexed, X Addressing mode turns into Absolute Indexed, X Addressing when operand is 0

rohanbafna, 2023-10-24:
When using an instruction with the Direct Page Indexed, X Addressing mode and the operand is 0, naken_asm will try to use the Absolute Indexed, X Addressing mode instead. Instructions like LDA which support both will assemble with the wrong addressing mode, while instructions like STY, which only support Direct Page Indexed, X, will fail to assemble.

Here's a sample input and output to explain what I mean:
```
        .65816
        lda.b 1,x
        lda.b 0,x
        sty.b 1,x
        sty.b 0,x
```
The assembler gives the error
> Error: No instruction found for addressing mode 8 at dpix.asm:5
when assembling this file.

The output listing is:
```
        .65816
        lda.b 1,x
0x0000: b5 01            lda 0x01,x                               cycles: ?

        lda.b 0,x
0x0002: bd 00 00         lda 0x0000,x                             cycles: ?

        sty.b 1,x
0x0005: 94 01            sty 0x01,x                               cycles: ?

        sty.b 0,x
0x0007:                  brk                                      cycles: ?
```
The first and third instructions assemble fine, but the second instruction uses the wrong addressing mode, and the fourth fails to assemble (because there's no opcode corresponding to STY with Absolute Indexed, X).

The culprit might be at [asm/65816.cpp:734](https://github.com/mikeakohn/naken_asm/blob/00eb2f10c85f05e5df0f38035f4cd361a2445c18/asm/65816.cpp#L734C37-L734C37).
```c++
        if (num == 0) { size = 16; }
```
This line executes before the assembler tests for simple indexed addressing modes. I assume it assumes `num == 0` only if a label was provided (so it can then be replaced with a 16-bit address), but if a literal `0` was written, then the assembler will incorrectly always assume a 16-bit operand.

mikeakohn, 2023-10-25:
Just wanted to let you know I saw this. I let Joe know about it...

JoeDavisson, 2023-10-25:
I think I know what to do but need a little time, thanks for reporting the problem.

mikeakohn, 2023-10-27:
Looks like this takes care of it.. let us know if you find anything else. Thanks!

closed 2023-10-27 by mikeakohn