mikeakohn/naken_asm #123: RISC-V: unexpected expansion of LI (load immediate)

ewost, 2023-11-02:
Hi, I'm evaluating naken_asm for a project and am quite impressed by the capabilities of this little assembler.
Today I found a (to me) strange behavior.
Please take a look at  a *.lst file: 
```
.riscv

   li x2, 0x12345678

0x00000000: 0x12346137 lui sp, 0x012346 
0x00000004: 0x67806113 ori sp, zero, 1656 (0x000678) 
```
Shouldn't LI expand to a LUI, ADDI sequence like so:
```
lui sp, 0x12345
addi sp, sp, 0x678

```
Thank you.

mikeakohn, 2023-11-02:
Cool to see someone's using the RISC-V part of the assembler :).

You caught me half way done with adding li... I know there are issues right now, I was planning on getting those taken care of in the next couple days.

ewost, 2023-11-02:
Ah, ok, thank you for clarifying. Then I will be patient and wait.
I am looking for a alternative to the GNU as. That beast is not really intended for handcrafted assembly. Seems to me that its main job is to assemble output from the C compiler.

mikeakohn, 2023-11-05:
This should be good for 32 bit RISC-V now.

mikeakohn, 2023-11-19:
64 bit RISC-V should be good now too.

closed 2023-11-19 by mikeakohn