mikeakohn/naken_asm #124: RISC-V feature request: compressed extension

ewost, 2023-11-02:
Well, that would not only be nice, but is really needed.
I work with a CH32V003 with 16K Flash. That's not much for a 32-bit ISA.
So every byte counts. Yes, that chip also has the E ISA, so only registers x0-x15 are implemented. But that should be no problem, just don't use regs x16 to x31. :-)
Are there plans to implement rvc?

mikeakohn, 2023-11-02:
If you want it, I'll add it :).

mikeakohn, 2023-11-02:
I should be able to get to it this weekend. There's some other extensions I want to add also .. I'm really interested in the vector extension... although I can't seem to get a board that supports it.

ewost, 2023-11-02:
> If you want it, I'll add it :).

Yes, please. That would be great. As a thank you for the quick support, maybe I can help with documentation. And testing, of course. :-)

mikeakohn, 2023-11-02:
Testing would be a really huge help.. and letting me know what features you need.

ewost, 2023-11-03:
Hello. Here is my wish list with notes:

- RV32C 
In GNU as, the engine automatically replaces 32-bit instructions with the compressed counterpart if possible. So you don't need to think about using compressed or not. It is hidden from you. But what if I don't want that in a part of my source code? So it has to be switchable. Oh man. That's complicated. I guess, I am fine with writing c.addi... if I want that instruction to be 16 bit.
- RV32E
If I understand correctly, the main (only?) difference between RV32I and RV32E is the lack of registers x16 to x31 in RV32E. It would be a nice feature if naken complains if those regs are used in RV32E mode. Could I live without it? Sure.
- Extension for compressed extension of CH32V003
Yesterday I learned that WCH has implemented a few additional Instructions to the compressed set like c.lb, c.lbu, c.lh, c.lhu and the store pendants. I am waiting for more information. Of course, this would also be nice to have, but the naken user can always use macros to fiddle together some stuff. :-)

Thank you for your support.

mikeakohn, 2023-11-10:
Got them all in.. probably buggy. I'll write some tests over the weekend to compare the new instructions against GNU as.

mikeakohn, 2023-11-19:
All instructions are being tested now except some jump/branch instructions (3 or 4 of them). It's still possible the permutations on immediate values could be off, but should be okay.

If a 32 bit instruction is added in a place where there should be a 16 bit instruction, the assembler will throw an error. It could auto-insert a c.nop, but I'm not sure that's a good idea.

mikeakohn, 2023-12-25:
This should be complete.

closed 2023-12-25 by mikeakohn