mikeakohn/naken_asm #125: MSP430 Simulator has incorrect Status Register flags on byte arithmetic operations

dlkeng, 2023-12-16:
The MSP430 Simulator appears to generate incorrect Status Register Z and C flags on several byte arithmetic operations. Word sized arithmetic operations appear to be correct.

More specifically, the **ADD.b, ADDC.b, CMP.b, SUB.b,** and **SUBC.b** operations do not always generate the correct Z and C status register flags in simulation.

This appears to be due to the masking operations always assuming 16-bit words, not accounting for 8-bit byte operations.
The following lines in naken_asm/simulate/msp430.cpp are suspect:
      - 860, 873, 884, 890, 900, 903, 913, 917

Attached is an assembly file used to confirm expected status operation of the byte instructions in question. This was tested/verified on a T.I. MSP430 LaunchPad using Code Composer Studio compiler/debugger.

[TestASM.asm.txt](https://github.com/mikeakohn/naken_asm/files/13693112/TestASM.asm.txt)

mikeakohn, 2023-12-17:
Sorry for a slow response, I was going to try to look into this yesterday, but ran out of time. I'll get on it.

And I appreciate you letting me know about this.. thanks!

mikeakohn, 2023-12-17:
That should do it. That test is really great. The only testing I've had for the simulators were in the Java Grinder project, and there are hardly any 8 bit operations in that. If I can get some time I'm going to add that to the tests here.

Thanks!

closed 2023-12-17 by mikeakohn