mikeakohn/naken_asm #132: FTBFS with only individual CPUs enabled

melak, 2024-10-08:
Current HEAD (080401), Mint 22 (gcc 13.2, glibc 2.39, readline 8.2), no `naken_asm` currently built or installed.
The following:
```bash
#!/usr/bin/env bash
# vim: tabstop=4 shiftwidth=4 expandtab colorcolumn=80 foldmethod=marker :     
                                                                               
CPUS="1802 4004 6502 65816 6800 6809 68hc08 680x0 8008 8048 8051 86000 arc arm"
CPUS="${CPUS} arm64 avr8 cell copper cp1610 dotnet dspic ebpf epiphany f100_l" 
CPUS="${CPUS} java f8 lc3 m8c mips msp430 padauk pdp8 pic14 pic18 powerpc"     
CPUS="${CPUS} propeller propeller2 emotion-engine riscv sh4 sparc stm8 super-fx" 
CPUS="${CPUS} sweet16 thumb tms340 tms1000 tms9900 unsp webasm xtensa z80"

for cpu in ${CPUS}
do
  exec > ../buildlogs/${cpu}.out
  exec 2>&1
  git clean -fdx
  ./configure --enable-${cpu} --include-path=`pwd`/include
  make
done
```
results in various failures for each CPU except for 680x0 (where the build runs to completion). The most common failure is
```
Now type: make

make -C build
make[1]: Entering directory '/tmp/naken/naken_asm/build'
g++ -c ../asm/common.cpp -o asm/common.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../asm/8008.cpp -o asm/8008.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../disasm/8008.cpp -o disasm/8008.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../common/add_bin.cpp -o common/add_bin.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../common/assembler.cpp -o common/assembler.o -Wall -DREADLINE -O3 -DNO_MSP430 -DENABLE_8008 -I..
../common/assembler.cpp: In member function ‘void AsmContext::init()’:
../common/assembler.cpp:93:10: error: no matching function for call to ‘AsmContext::set_cpu(AsmContext*, int)’
   93 |   set_cpu(this, 0);
      |   ~~~~~~~^~~~~~~~~
In file included from ../common/add_bin.h:17,
                 from ../asm/common.h:17,
                 from ../common/assembler.cpp:16:
../common/assembler.h:44:8: note: candidate: ‘void AsmContext::set_cpu(int)’
   44 |   void set_cpu(int index);
      |        ^~~~~~~
../common/assembler.h:44:8: note:   candidate expects 1 argument, 2 provided
../common/assembler.h:45:7: note: candidate: ‘int AsmContext::set_cpu(const char*)’
   45 |   int set_cpu(const char *name);
      |       ^~~~~~~
../common/assembler.h:45:7: note:   candidate expects 1 argument, 2 provided
make[1]: *** [Makefile:33: common/assembler.o] Error 1
make[1]: Leaving directory '/tmp/naken/naken_asm/build'
make: *** [Makefile:12: default] Error 2
```
but then there's a different (link-time) failure with the msp430 target.
The complete output is attached as a tarball here: 
[buildlogs.tar.gz](https://github.com/user-attachments/files/17289963/buildlogs.tar.gz)

mikeakohn, 2024-10-08:
I'm not quite understanding what is wrong. What are you doing exactly?

melak, 2024-10-08:
This:
```
bethan:/tmp/naken/naken_asm(master)$ ./configure --enable-z80

Configuration:
  Install to: /usr/local
            : /usr/local/bin, /usr/local/share/naken_asm/include
    Compiler: gcc, g++
     LDFLAGS:  -s
UTIL LDFLAGS:  -lreadline
      CFLAGS: -Wall -DREADLINE -O3 -DNO_MSP430
     DEFINES:  -DENABLE_Z80
INCLUDE_PATH: /usr/local/share/naken_asm/include
        BINS: naken_asm, naken_util

Now type: make

bethan:/tmp/naken/naken_asm(master)$ make
make -C build
make[1]: Entering directory '/tmp/naken/naken_asm/build'
g++ -c ../asm/common.cpp -o asm/common.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../asm/z80.cpp -o asm/z80.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../disasm/z80.cpp -o disasm/z80.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../common/add_bin.cpp -o common/add_bin.o -Wall -DREADLINE -O3 -DNO_MSP430 -I..
g++ -c ../common/assembler.cpp -o common/assembler.o -Wall -DREADLINE -O3 -DNO_MSP430 -DENABLE_Z80 -I..
../common/assembler.cpp: In member function ‘void AsmContext::init()’:
../common/assembler.cpp:93:10: error: no matching function for call to ‘AsmContext::set_cpu(AsmContext*, int)’
   93 |   set_cpu(this, 0);
      |   ~~~~~~~^~~~~~~~~
In file included from ../common/add_bin.h:17,
                 from ../asm/common.h:17,
                 from ../common/assembler.cpp:16:
../common/assembler.h:44:8: note: candidate: ‘void AsmContext::set_cpu(int)’
   44 |   void set_cpu(int index);
      |        ^~~~~~~
../common/assembler.h:44:8: note:   candidate expects 1 argument, 2 provided
../common/assembler.h:45:7: note: candidate: ‘int AsmContext::set_cpu(const char*)’
   45 |   int set_cpu(const char *name);
      |       ^~~~~~~
../common/assembler.h:45:7: note:   candidate expects 1 argument, 2 provided
make[1]: *** [Makefile:33: common/assembler.o] Error 1
make[1]: Leaving directory '/tmp/naken/naken_asm/build'
make: *** [Makefile:12: default] Error 2
```

melak, 2024-10-08:
I think that's just masking whatever the actual issue is.
I'm no C++ guy by any stretch of the imagination, but
```
diff --git a/common/UtilContext.cpp b/common/UtilContext.cpp
index e96b3e1f..b81bedf5 100644
--- a/common/UtilContext.cpp
+++ b/common/UtilContext.cpp
@@ -83,7 +83,7 @@ void util_init(UtilContext *util_context)
   util_context->alignment = 1;
 #else
   util_context->disasm_range = cpu_list[0].disasm_range;
-  util_context->simulate = SimulateNull:init(&util_context->memory);
+  util_context->simulate = SimulateNull::init(&util_context->memory);
   util_context->flags = cpu_list[0].flags;
   util_context->bytes_per_address = cpu_list[0].bytes_per_address;
   util_context->alignment = cpu_list[0].alignment;
diff --git a/common/assembler.cpp b/common/assembler.cpp
index 1da82fff..72701019 100644
--- a/common/assembler.cpp
+++ b/common/assembler.cpp
@@ -90,7 +90,7 @@ void AsmContext::init()
   list_output = list_output_msp430;
   cpu_list_index = -1;
 #else
-  set_cpu(this, 0);
+  set_cpu(0);
 #endif
   address = 0;
   instruction_count = 0;
```
this almost looks like it would make sense?

melak, 2024-10-08:
Sprinkle with
```
diff --git a/configure b/configure
index 6ea2816e..ad846bf4 100755
--- a/configure
+++ b/configure
@@ -76,7 +76,7 @@ ASM_OBJS="common.o"
 DISASM_OBJS=""
 TABLE_OBJS=""
 UTIL_OBJS="UtilContext.o util_disasm.o util_sim.o"
-SIM_OBJS="null.o"
+SIM_OBJS="Simulate.o null.o"
 NO_MSP430="-DNO_MSP430"
 DFLAGS_ALL=""
 
```
and it's showing signs of building as originally intended (save for 65816, arc, ebpf, lc3 and riscv, but at least we are this much closer).

mikeakohn, 2024-10-08:
I added your fixes and also removed the options to select individual CPUs from the ./configure script.

This "feature" is causing too many problems and I'm guessing nobody else uses it so in the next couple months or so I will remove the option completely.

closed 2024-10-08 by mikeakohn