mikeakohn/naken_asm #134: Z80 assembler generates dual instructions for LD (HL),r

dlkeng, 2024-10-08:
The _naken_util_ Z80 assembler generates instructions for the **LD (HL),r** instruction twice, back-to-back. This can be seen in the below trace:
```
stopped> print 0x1000-0x1008
0x1000: 00 00 00 00 00 00 00 00 00
stopped> asm 0x1000
asm> ld (hl),b
asm>
Assembling to 0x1000
stopped> print 0x1000-0x1008
0x1000: 70 70 00 00 00 00 00 00 00
stopped> disasm 0x1000-0x1004

Addr    Opcode Instruction
------- ------ ----------------------
0x1000: 70        ld (hl),b
0x1001: 70        ld (hl),b
0x1002: 00        nop
0x1003: 00        nop
0x1004: 00        nop
stopped>
```

mikeakohn, 2024-10-09:
I'm not seeing this:

```
stopped> asm 0x1000
asm> ld (hl),b
asm> 
Assembling to 0x1000
code: ld (hl),b

stopped> disasm 0x1000-0x1004

Addr    Opcode Instruction                              Cycles
------- ------ ----------------------------------       ------
0x1000: 70        ld (hl),b                                4
0x1001: 00        nop                                      4
0x1002: 00        nop                                      4
0x1003: 00        nop                                      4
0x1004: 00        nop                                      4
stopped> 

```

I tried it on macOS also and not seeing it. I tested it directly after bringing up naken_util and doing nothing else. Anything other info on what you're doing?

dlkeng, 2024-10-10:
OK, I downloaded the naken_asm-2024-02-10.zip (Windows Binary) from your website and it does not show the problem either.

So, I guess it must have been something I've done in adding the Z80 simulator. I tried tracing into the running assembler source code, but got lost in all of the token processing going on there.

I had to change the position of a few items in table_z80[] in table/z80.cpp to have more-specific masked items to be listed before a less-specific masked it in order for the proper opcode could be found. Replacing the table_z80[] with the original version fixed the problem, but broke the Z80 simulator I'm working on.

I guess I'm going to have to investigate more to figure out what is happening!

closed 2024-10-10 by dlkeng